bztsrc/raspi3-tutorial #1: Make Fails at Line 851

btauro, 2018-01-12:
Hey,
Excited to work with the tutorials but when i run the command make -j4 inside binutil 2.29
Get the following error
make[2]: Leaving directory '/home/brian/binutils-2.29/libiberty'
make[1]: Leaving directory '/home/brian/binutils-2.29'
make: *** [Makefile:851: all] Error 2

Tried in fedora and ubuntu same error.

bztsrc, 2018-01-12:
Hi,

Not sure as you haven't provided enough output, but seems you have problems with the libiberty or with one of it's dependencies. Read your full output carefully, I'm 100% sure the reason for the error is printed there somewhere. Try to re-install libiberty-dev or ask on Fedora and/or Ubuntu forums "How to compile a cross-compiler". I don't use either of those dists, and I'm afraid I cannot help with this more than point you in the right direction.

I want to clarify this tutorial is not about cross-compilers, that's why it's numbered zeroth.

Good luck,
bzt

closed 2018-01-12 by bztsrc

initard, 2021-06-15:
Hi, I had this same issue and I want to share how I fixed it. I hope this is helpful to someone in the future. 

The issue had to do with no having all the packages needed installed. See this https://wiki.osdev.org/GCC_Cross-Compiler for knowing what to install. This should at least get you past the Makefile:851 error. 

As a side note, doing `make -j1` instead of `-j4` also worked for me for binutils

Good luck