bztsrc/raspi3-tutorial #5: Not showing anything on serial port

btauro, 2018-01-17:
The screen is blank when using terminal emulator with Rapberry Pi3. Not sure why this is happening.
Previous Tutorials with UART 0 have worked.But this tutorial for some reason does not print anything to the console,I suppose something to do with start.S.Not Sure.Could you help me with this

bztsrc, 2018-01-17:
Okay, first thing first, which tutorial are you talking about?
If I understand right, the UART1 tutorials (1-4) were okay, and there was at least one (or more?) with UART0 (>4) that worked?

btauro, 2018-01-17:
Hey,
1] All tutorials  before 0F_executionlevel have worked.
2]It is minicom terminal emulator
3] Yes it is set to 115200,8n1
4]I am using the recent commit

bztsrc, 2018-01-17:
Thank you for the feedback, I'll look into that!

bztsrc, 2018-01-18:
Hi btauro,

Bad news, I'm afraid. It turned out that my first test failed because of a sync error. I've compared the code in the tutorial with my original code, and there was only one difference, setting the virtual mpidr register. I've tested with and without it, both worked.
Since I'm unable to reproduce your error, I'm not sure what the problem could be. I've attached a screenshot, first I've tried my original code, then the tutorial with vmpidr code, then the unmodified tutorial. Finally I've tested the unmodified tutorial again, just to be sure.

Cheers,
bzt
[image]

closed 2018-01-18 by bztsrc

btauro, 2018-01-19:
Could you share your config.txt with me i guess this may be the cause.

This is my config.txt
arm_control=0x200               
kernel_old=1
disable_commandline_tags=1
My qemu Output
[image]

Unfortunately not working on Raspberry Pi3 for some reason

Do i require any additional files on my sd card.These are the files on my SD card
bootcode.bin
config.txt
start.elf
kernel8.img
fixup.dat

[image]