bztsrc/raspi3-tutorial #6: The Current El level in Raspberry Pi 3 is 00000003

btauro, 2018-01-20:
I have commented the change of execution levels code and i realized my raspberry pi3  CurrentEl starts at 0000003
Do you think this is the cause for no output 
when i try including the chamge of  execution levels with the below code i get no output.So could this be the cause? 

```
mov     x2, #0x3c4
    msr     spsr_el2, x2
    adr     x2, 5f
    msr     elr_el2, x2
    eret

```

bztsrc, 2018-01-20:
Well, if you really run at EL3, then setting up a return from EL2 won't do any good. You have to use EL3 registers instead. Although I'm very sceptical about this, because RPi was made to boot Linux primarily, and Linux kernel [expects EL2](https://github.com/torvalds/linux/blob/master/arch/arm64/kernel/head.S#L580) at most.

bzt

bztsrc, 2018-01-20:
Anyway, I've added code for switching to EL2 from EL3, let me know if it works! (I can't test it, as my hardware runs kernel8.img at EL2).

bzt

btauro, 2018-01-20:
Hey,

Thank you so much for helping me.

But  I still did not get serial output.

But if i remove the below lines my current El changed and now my Current EL is: 00000002 

```
mov     x2, #0x3c4
msr     spsr_el2, x2
adr     x2, 5f
msr     elr_el2, x2
eret
```

btauro

btauro, 2018-01-21:
Finally with Gods grace got it working
adding these two lines solved the issue

```
//set up el1 toAARCH64
mov     x1,#0x80000000
msr     hcr_el2,x1
```
Thank you so much

bztsrc, 2018-01-21:
Wtf? That "mov" is not a valid instruction. I suppose since the lower bits are all zero, your assembler automatically converts it into a "movk x1, #0x8000, lsl #16"? Also, you should not clobber x1 register as it holds the address of the stack.
But you were right about one thing, hcr_el2 should be set to tell the CPU EL1 is running in AArch64 mode. So I've added it to the CurrentEL tutorial (the others already had it, I just added comments in their source).

It seems you have found another thing that differs in qemu and in real hardware. I've already noticed that qemu clears hcr_el2 bit 1, while on real hardware it is reserved to be set. Now we know it does not handle bit 31 correctly either.

Thank you very much for your feedback,
bzt

closed 2018-01-21 by bztsrc