bztsrc/raspi3-tutorial #8: Unable to find a usable island

JayyCee, 2018-02-13:
Does anyone know how to resolve this error? ( I'm using Ubuntu 16.04)

After a successful  "sudo make install" inside binutils-2.29, then I stared build gcc, and got this error:

./configure <options...>
...
checking for isl 0.15 or later... no
required isl version is 0.15 or later
configure: error: Unable to find a usable isl.  See config.log for details.
jchen@ubt-naas ~/joe-git-workspace/raspi3-bare-metal/tool-chain/gcc-7.2.0

JayyCee, 2018-02-14:
It turned out that, it seems to me,  the following  libraries need to be installed on the host (Ubuntu 16.04 in my case) before "./configure" for the gcc-src to be invoked:

libmpc-dev
libmpfr-dev
libisl-dev

After this, "./configure" completes without any error.

bztsrc, 2018-02-14:
Yes, if you have the option to use your system's packaging manager you should do that instead of downloading the source directly. Packaged sources are often patched to the needs of that specific system.

closed 2018-02-14 by bztsrc

vp080808, 2018-04-15:
Hi all,

My system is OSX 10.13.4 andI had a bunch of hurdles to go through to get everything working.
One of the major issues is that macports installs a libiconv which has different symbols from the Xcode 9.3 version. E.g.
```
vp-lion:/ % nm /usr/lib/libiconv.2.dylib | grep iconv
00000000000f1d80 S ___iconv_2VersionNumber
00000000000f1d50 S ___iconv_2VersionString
00000000000f3750 D __libiconv_version
0000000000002b15 T _iconv
0000000000002e33 T _iconv_canonicalize
0000000000002b37 T _iconv_close
000000000000188d T _iconv_open
0000000000002b44 T _iconvctl
0000000000002c40 T _iconvlist
0000000000014f10 T _libiconv_relocate
00000000000f3760 b _libiconv_relocate.initialized
0000000000014e57 T _libiconv_set_relocation_prefix
vp-lion:/ % nm /opt/local/lib/libiconv.2.dylib | grep iconv
00000000000f67d0 D __libiconv_version
000000000000374a T _iconv_canonicalize
0000000000002f72 T _libiconv
0000000000002f94 T _libiconv_close
0000000000001c5c T _libiconv_open
0000000000002fa1 T _libiconv_open_into
0000000000016c79 t _libiconv_relocate
0000000000016bc0 T _libiconv_set_relocation_prefix
000000000000345c T _libiconvctl
0000000000003558 T _libiconvlist
```

So to get everything working I had to follow this procedure:

_a) download ONLY binutils and gcc. The others should be installed using a packages manager (e.g. macports)._

_b) configure binutils with:_

**cd binutils*
sh configure --prefix=/usr/local/cross-compiler --target=aarch64-elf --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-system-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-plugin --disable-linker-build-id --enable-lto --enable-install-libiberty --with-linker-hash-style=gnu --with-gnu-ld --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-gnu-unique-object**

(note the --with-system-isl option which allows you to use the installed version of isl)

**make -j4**
        (j4 means 4 cores)
**sudo make install**

        if you get errors such as 
clang: error: unsupported option '-print-multi-os-directory'
        do not worry, ignore them

        The installation places files at: /usr/local/cross-compiler
        So be sure to add this directory in your path to execute the new commands
**set path= ( /usr/local/cross-compiler/bin $path )**

_c) configure gcc with:_

**setenv LIBRARY_PATH=/usr/lib:/opt/local/lib:/usr/local/lib**
        (or if you use /bin/sh export LIBRARY_PATH=/usr/lib:/opt/local/lib:/usr/local/lib)
**cd ../gcc*
sh configure --disable-nls USE_NLS=no --prefix=/usr/local/cross-compiler --target=aarch64-elf --enable-languages=c --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-plugin --disable-linker-build-id --enable-lto --enable-install-libiberty --with-linker-hash-style=gnu --with-gnu-ld --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-gnu-unique-object --with-gmp-include=/opt/local/include --with-gmp-lib=/opt/local/lib**

(note the use of options --with-gmp-include=/opt/local/include --with-gmp-lib=/opt/local/lib to use the installed version of the gmp)

**make -j4 all-gcc**

        if this fails with some iconv related error:

 **cd host-x86_64-apple-darwin17.5.0/gcc**
        edit the Makefile to change the definition of LIBICONV to
LIBICONV = /usr/lib/libiconv.dylib

**make clean
then make -j4
cd ../..
sudo make install-gcc**

You are done.

**vp