bztsrc/raspi3-tutorial #9: 00_crosscompiler fixes

solskogen, 2018-02-20:
binutils does not use isl, so there's no reason to link it. Yes, the configure script searches for it, but it's not used. binutils and gcc share the same ./configure script.

gcc does not use cloog anymore. And there's no need to specify "--with-isl" as long as the symlink is in the gcc source directory.

bztsrc, 2018-02-20:
Dear solskogen,

Thank you for your comments! For me binutils is still requiring isl (not sure why, bit it does). The configure script for binutils and gcc are different, I assure you. I know because I had to patch gcc's script to make the -fvisibility flag available in gcc (it is possible that ideally they should be the same, but in reality they are two different files with lots of differences, I've just ran a diff). About cloog and the --with-isl flag I'm not sure. It's quite some time I've wrote this, it is possible indeed that gcc changed since. I'll check it again and update the tutorial, thanks again!

solskogen, 2018-02-20:
Sometimes the configure script for gcc and binutils differ, but you can actually build binutils and gcc together if you symlink bfd binutils gas gold ld opcodes from binutils into gcc source directory. 

"grep -r with-isl" in the binutils source directory, and you'll see that nothing uses isl :-)

bztsrc, 2018-02-21:
So they sometimes are the same, but not in all configurations.
I believe you that isl is not referenced, but regardless my binutils configure script fails if it's missing...
So for now I'll leave the tutorial as is. This is not a gcc cross-compiler tutorial, just a cheat sheet.

closed 2018-02-21 by bztsrc