bztsrc/raspi3-tutorial #11: Troubleshooting

milanvidakovic, 2018-03-25:
Hi,
Thank you for this great code. Unfortunately, I am not able to run any of the examples successfully. Here is what I have tried:
1. format the sd card as FAT32
2. copy bootcode.bin and start.elf onto the sd card
3. copy kernel8.img from an example which should output to the serial card, right to the sd card
4. put the sd card into the rpi3
5. power on

Now, from that moment on, I don't get anything on the serial port (I have a working serial-to-usb dongle, which does work properly). 
How can I troubleshoot this?

Thanks in advance!

bztsrc, 2018-03-26:
Hello,

Debugging boot procedure on a real hardware is not easy, requires a deep knowledge of the process and a big deal of experience.

I'd suggest to check these:
1. is the partitioning table MBR (and not GPT) with partition type 0xC? How big is your partition? I've found that not all partition sizes are supported by the fw. (In short, can the fw find the files at all?)
2. try to download the other firmware files too, most notably fixup.dat
3. if you use a different kernel image, does it work? (so narrow the problem to either GPU booting (bootcode.bin, start.elf) or ARM (kernelX.img; a led blinker kernel7.img will do))
4. are you sure your board supports AArch64? (so is it RPi3 and not RPi2B+ by any chance?)
5. if you have a config.txt on the SD card, remove it, as my tutorials were intended to boot with default configuration. They won't work if you have some special set up, like kernel_old.

This is not a complete list, just a few thoughts what you should check. The most important thing is to narrow down whether the problem is with your card (partition unreadable by the fw), with the GPU boot stage, or with executing the ARM kernel. Also you could try to download a raspbian lite image and replace the kernel in it (with that you can rule out the first two possibilities for sure).

I strongly suggest to start with a known to be working image (like raspbian), remove kernel*.img and config.txt and build your testing environment up from there.

Good luck!
bzt

milanvidakovic, 2018-03-26:
Hello,
Thank you for your detailed answer. I have purchased a new 16GB Kingston class 10 sd card and I have installed raspbian. It booted without problems. Then I have deleted almost everything, leaving bootcode.bin and start.elf. Then I have copied your simple uart echo example as kernel8.img, placed the sd card into the RPI3 (yes I do have RPI3) and everything was OK.

Now I need one more thing (for now) - I would like to use the your raspbootin64 image to reduce sd card wear and send executables via serial cable. However, my development environment is Windows. You have left a link to a windows-based raspbootin-server.c. This information is rather foggy to me. What can I do with that file? To me it doesn't appear as a client-side program, which could be used to send executable files via serial connection to your raspbootin64.  
Github dwelch67 user has created similar environment as yours, but it uses xmodem protocol to accept files to be executed. On Windows, I can find several xmodem clients which I could use to send files via serial cable. Could you help regarding this matter? 
Best regards,
Milan

milanvidakovic, 2018-03-27:
Hello,
let  me reply to my own question. Now that I can boot from any kernel8.img file, I have decided to write a Java client for your Raspbootin loader. 
I have written an application in Java which sends arbitrary kernel file to the your Raspbootin loader. I am getting all the right answers from the loader, including the 'OK' at the end, but it seems that the loaded kernel file is not working. 
Just to check: your Raspbootin expects a kernel8.img file from the examples to be sent?

[Raspbootin64Client.zip](https://github.com/bztsrc/raspi3-tutorial/files/1850340/Raspbootin64Client.zip)

milanvidakovic, 2018-03-27:
Hello,
let me reply to my own question. Now that I can boot from any kernel8.img file, I have decided to write a Java client for your Raspbootin loader.
I have written an application in Java which sends arbitrary kernel file to the your Raspbootin loader. I am getting all the right answers from the loader, including the 'OK' at the end, but it seems that the loaded kernel file is not working.
Just to check: your Raspbootin expects a kernel8.img file from the examples to be sent?
[Raspbootin64Client.zip](https://github.com/bztsrc/raspi3-tutorial/files/1850345/Raspbootin64Client.zip)

bztsrc, 2018-03-27:
Hi,
That windows code was not written by me, and I have never tried it. But you should be able to compile it on win and use it to send your kernel just like with the original rasbootcom utility. If you have written another app, that's great!
Yes, the kernel file should be the same kernel8.img you would load from sd card if you were using sd card and not serial.
Please give me a few days, somebody else has reported problems with the loader, although I was unable to reproduce the error so far. I'll do more through tests and get back to you.

bztsrc, 2018-03-27:
I've tested it, now works for me. Please check it out.

milanvidakovic, 2018-03-27:
Hi,
Thank you. I am waiting for your response :)
In a meantime, just to inform you how far I have reached: loader receives the kernel8.img via serial and then jumps to the 0x80000 address. That is the point where uploaded program fails. I don't know what happens there, because that step should be straight forward. Unfortunately, nothing happens at that moment, so I don't know if the uploaded image gets corrupted during upload, or something else fails.
Best regards,
Milan

bztsrc, 2018-03-27:
Yeah, I agree, that step should be straight forward, but it's not. I suppose it's a gcc bug, because it does not jump to the address 0x80000 as one would expect. First I've wanted to use a relocation calculated by hand, but that seemed way too ugly. So instead I've decided to use a fake function return to jump to the absolute address of the new kernel, which is 4 bytes longer, but at least works :-)

milanvidakovic, 2018-03-27:
Hello, 
It works! :) Thank you!
Now I will go through all the examples.
I have changed the Java client so it works a bit more stable. I plan to put the source to my github repository, but in any case, if you don't mind, here is the whole Eclipse project. Works on Windows, and should work on any OS with Java.
Best regards,
Milan
[Raspbootin64Client.zip](https://github.com/bztsrc/raspi3-tutorial/files/1853322/Raspbootin64Client.zip)

bztsrc, 2018-03-27:
Good to hear!
Please let me know when you've created a github repo for your Java kernel sender, I'd like to link it if you don't mind.

Cheers,
bzt

closed 2018-03-27 by bztsrc

milanvidakovic, 2018-03-28:
Hello,
I have created a repo on the github:
https://github.com/milanvidakovic/Raspbootin64Client

Thanks again for the great software!
Best regards,
Milan