bztsrc/raspi3-tutorial #12: Clock Interrupt

milanvidakovic, 2018-03-28:
Hi,
is there any support for the clock interrupt? I have read somewhere that RPI has 1MHz clock-driven interrupt which would help me greatly since I need quite precise timing for the signal generation using GPIO. So far, using your 07_delays example, timing gets corrupted when approaching microseconds. For example, I need to create a pulse 3.8 microseconds long, for each 31.7 microseconds (horizontal sync VGA signal). Using all three variants of wait (cpu cycle, cpu wait and hw wait) got me very unstable timing.
Best regards,
Milan
P.S. I am working on the GUI Raspbootin64 loader in Java. I will commit to my github repo when finished.

bztsrc, 2018-03-29:
Hi,
Yes, there is. More actually. You can use the ARM or the BCM System Timer too. I'm planning to do some timer interrupt tutorials, but I haven't got the time so far. Until then, I suggest to take a look at the other tutorials I've linked on the front page, they have example codes. Also you should definitely take a look at Circle64: https://github.com/rsta2/circle64/blob/master/lib/timer.cpp It's using the BCM System Timer to generate microsec precision interrupts for kernel ticks.
Good to hear, I've liked your repo :-)

closed 2018-03-29 by bztsrc

milanvidakovic, 2018-03-29:
Hello,
Thank you, I will look. C++ is one of my least favorite languages, but I will give it a look, as well as on other links you mentioned.
Best regards,
Milan

lizhuohua, 2018-05-10:
@bztsrc It will be fun if you do some timer interrupt tutorials. Really look forward to it.
Actually I tried to use the system timer to generate interrupts. It works well on hardware but the qemu doesn't seem to simulate it. Do you have any suggestions about simulating timer interrupts in qemu? Thanks a lot.

bztsrc, 2018-05-11:
@lizhuohua: yes, the system timer (which is a BCM peripheral) is not emulated in qemu. There's an alternative way to generate timer interrupts purely with ARM, I suggest to look into that (see DDI0487 doc). I'm planning to write an interrupt and clock tutorial for quite some time now, unfortunatelly haven't had the time so far.