bztsrc/raspi3-tutorial #13: uart.c clock mailbox request returns error

andre-richter, 2018-03-31:
Hi Zoltan,

while implementing your code in another language, I checked the response of the uart.c init mailbox call, and it returns error `0x80000001: error parsing request buffer (partial response)`.
I think the field `skip setting turbo` is missing ([reference link here](https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface)).

I was too lazy too look up what a meaningful value is, but setting it with zero works for me for now:
https://github.com/andre-richter/rust-raspi3-tutorial/blob/master/05_uart0/src/uart.rs#L69-L76

Best Regards,
Andre

bztsrc, 2018-04-01:
Hello,

Thanks for the feedback. All that I can tell is, it works for me. Without that mailbox call I got nothing but jibrish, meaning the baud rate is not valid. With the call everything is fine, so I do not bother about the return status code (besides, it's not that we can do anything about it in the unlikely case if it fails). The errorcode btw suggests you've used another tag with bad arguments. Are you sure you've set up the mailbox array properly? Maybe invalid mailbox size or missing last tag marker?
Anyway I'll check it and see about that turbo flag you've mentioned, just to be sure everything is ok.

Thanks for your report.
bzt

andre-richter, 2018-04-01:
Yeah I can confirm that it works despite the error, and the call is effective and needed. I tested it with your C code as well, return code is 0x8000001 which means partial error or something like that. I just wanted to report for the sake of completeness :)

bztsrc, 2018-04-01:
I appreciate that, thanks! I'm guessing I'd have to add a zero int to the array right before the MBOX_TAG_LAST and change the first element to 9*4. That should make it return a full success. Since the code is working as is, I'll close this issue now, but I'll fix that call.

closed 2018-04-01 by bztsrc

bztsrc, 2018-11-10:
I'd like to apologize, because I've totally forgotten about your issue. Now it was reported by another user, so debugged it again (my bad), and it's been fixed.