bztsrc/raspi3-tutorial #14: Cannot find the raspi3 emulation capability in QEMU

OllieLollie1, 2018-04-10:
I have compiled the aarch46-softmmu target for QEMU however I believe that I haven't found the latest source because I have been unable to find any raspberry pi 3 emulation capability. I got the source from [](https://github.com/qemu/QEMU). would you please help me and others achieve raspberry pi 3 emulation

bztsrc, 2018-04-10:
Hi,

Well, that source is definitely has the patch: https://github.com/qemu/qemu/blob/master/hw/arm/raspi.c#L244
If you list the supported machines on the command line, do you see raspi2 and raspi3? If not, then first try to use the flags from my qemu-raspi3 repo's readme when you compile qemu. (But I've used only my repo, haven't checked the official qemu source as I supposed they can apply a simple patch without probs). Second, and worst case, overwrite the files from my repo and follow the instructions there, that's tested and known to be working.

OllieLollie1, 2018-04-10:
I cannot use your repo because it doesn't compile with qemu there are major differences in the files between the two repos. When I read the qemu source it looks like it has raspi3 emulation capability however when I run qemu-system-aarch64 -M help  it doesn't list raspi3 it only lists raspi2.

bztsrc, 2018-04-11:
Sad to hear :-(

Are you sure you're running the compiled version and not the binary distribution? Try "./qemu-system-aarch64 -M help" in the source directory (note the dot and slash at the beginning).

I'm afraid there's nothing more I can do, you should open a ticket on qemu repo's issues. Let's hope they'll fix it soon, or they can point you in the right direction. In the meantime I can recommend to use an older version of qemu, 2.11.50 where my patch works for sure.

Good luck,
bzt

closed 2018-04-11 by bztsrc

OllieLollie1, 2018-04-11:
I am so sorry I have never made that mistake before I neglected to add the ./ at the beginning of the command :)