bztsrc/raspi3-tutorial #16: How to find EOF

OllieLollie1, 2018-04-26:
Is it possible to read the metadata to find the end of file?

bztsrc, 2018-04-26:
Absolutely. I already read the file metadata in https://github.com/bztsrc/raspi3-tutorial/blob/master/0D_readfile/fat.c#L135 . The fatdir_t struct also has a file size member.

In order to detect EOF either you would check end of cluster chain marker (as in the tutorial), or substract the number of bytes read from size in every iteration until it's zero. I have an example for the latter in the bootloader I wrote: https://github.com/bztsrc/bootboot/blob/master/aarch64-rpi/bootboot.c#L1180 I have an extra check in line 1182 which guarantees no more than the remaining bytes is read, therefore the loop variable will never overflow (or in other words became negative).

closed 2018-04-26 by bztsrc

OllieLollie1, 2018-04-26:
Because I am doing the readfile tutorial I was able to check if the char == 0 if that was true it was the end of the file

bztsrc, 2018-04-27:
Yeah, because it's not a good solution. There's no guarantee that text files are terminated by a zero on disk, and binary files (like a png or elf) most certainly will contain zero characters in the midst.
I recommend to compare the bytes read to the file size, or check the end of cluster chain marker in order to detect EOF properly, as I've said.