bztsrc/raspi3-tutorial #17: List directory function not listing everything

OllieLollie1, 2018-04-26:
Is there a reason for the fat_listdirectory() function to not list everything I have been trying to read files with similar names (20 chars or so with a number at the end) is there a way to achieve this?

bztsrc, 2018-04-27:
Yes, the reason is to exclude long file name records and deleted files. Normal FAT filenames are 11 bytes long (8+3), supporting more is subject to M$ patents. Using them would require to pay a license fee, clearly not an option for a free open source tutorial.

But in a nutshell, originally there's a one-to-one relation in directory records to clusters, meaning every record belong to a different starting cluster number. With LFN, you have several records related to the same cluster number. One of them is a generated short name (most likely with a ~1 at the end) compatible to the original FAT spec. The other records starts with a special character, have a sequence number, and the remaining space in the record is filled up by 2 bytes unicode characters. Concatenating those characters in the order specified by the sequence number gives you the full long file name for that starting cluster.
https://www.codeguru.com/cpp/cpp/cpp_mfc/files/article.php/c13907/Long-File-Name-LFN-Entries-in-the-FAT-Root-Directory-of-Floppy-Disks.htm
http://www.osdever.net/documents/LongFileName.pdf
http://www.osnews.com/story/21766/Linux_Kernel_Patch_Works_Around_Microsoft_s_FAT_Patents

closed 2018-04-27 by bztsrc

OllieLollie1, 2018-05-02:
If I do implement it does that violate microsoft patents?

bztsrc, 2018-05-02:
I already gave you the answer to that.

http://www.osnews.com/story/21766/Linux_Kernel_Patch_Works_Around_Microsoft_s_FAT_Patents
"Andrew Tridgell, of SAMBA fame, has posted a patch on the LKML that does just that. The new patch offers support for long file names without infringing any of Microsoft's patents. This isn't just an empty claim; several patent lawyers with expertise in this area have investigated the patch and concluded that it does not infringe the patents."

So yes it is possible.