bztsrc/raspi3-tutorial #19: Random range erroneous.

fluxrider, 2018-05-03:
The formula should be *RNG_DATA % (max - min + 1) + min, because the numbers are integers.

https://github.com/bztsrc/raspi3-tutorial/blob/173e3090a6127f5fa92e11c80afbb9a3a49f6d02/06_random/rand.c#L52

bztsrc, 2018-05-04:
Thank you for your comment, however it's not about integers.

Your formula could return max, which I prefer not to. I like to have rand(0,10) return ten possible numbers: 0,1,2,3,4,5,6,7,8,9 and not the eleventh 10. The only problem here is if you incorrectly pass the same min and max value to the function, which makes no sense (but won't cause trouble as AArch64 does not have a division by zero exception). But in a real life application you'd have to check the arguments anyway, also if min is not greater than max. That is out of the scope of this tutorial. This is about how to access RNG, and not about programming best practices.

Cheers,
bzt

closed 2018-05-04 by bztsrc

fluxrider, 2018-05-04:
Thanks for the comment. I am used to max being included. So that rand(1, 6)
is a 6-sides dice.

No worries,
David