bztsrc/raspi3-tutorial #26: Black screen - no display

falkor2k15, 2018-05-20:
We can't get raspi3 to work on the newest Qemu - just a black screen - nothing seems to display regardless of what kind of FAT32 storage containing boot files I add to this command line:
qemu-system-aarch64 -M raspi3 -m 1G -bios kernel8.img

bztsrc, 2018-05-24:
You are telling qemu to directly load the kernel with the "-kernel" command line option (and not "-bios"). As you haven't specified FAT32 storage with "-drive" at all, it is not used. See the README.md for details.
https://github.com/bztsrc/raspi3-tutorial#emulation

It is absolutely normal to have a blank screen, only the 09_framebuffer and 0A_pcscreenfont displays something. All the other tutorials writes to the serial line, either to UART0 or UART1. Therefore you have to tell qemu to display the serial output on your terminal with "-serial stdio" otherwise you see nothing.

closed 2018-05-24 by bztsrc

xubenji, 2021-01-26:
In 03_uart1, I emulate the os in qemu with this insturction: qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial stdio
I added -serial stdio, but I still can't see anything in qemu.

bztsrc, 2021-01-30:
Try running `make run`. There's a correct qemu invocation in all [Makefile](https://github.com/bztsrc/raspi3-tutorial/blob/master/03_uart1/Makefile.gcc). For 03_uart1, that is
```
qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial null -serial stdio
```

Cheers,
bzt