bztsrc/raspi3-tutorial #27: Virtual Memory example deeply floored

LdB-ECM, 2018-06-11:
The virtual memory example is deeply flawed

First there is no such thing as undocumented TTBR_ENABLE, read again what x is on ttbr0_el1 and ttbr1_el1

Specifically this:
Bits [x-1:0] are res0.x is based on the value of TCR_EL1.T0SZ, the stage of translation, and the memory translation granule size.  If bits [x-1:0] are not all zero, this is a misaligned Translation Table Base Address. Its effects are constrained unpredictable, where bits [x-1:0] are treated as if all the bits are zero. The value read back from those bits is the value written.

You are using a 4k granual so bit0 to bit 11 are supposed to be zero, you were supposed to make sure you align the table to 4k to guarantee that.

Your  TTBR_ENABLE being set  to 1 makes a  misaligned Translation Table Base Address.

You set invalid flags in the L1 block table entries 

// TTBR0, identity L1
paging[0]=(unsigned long)((unsigned char*)&_end+2*PAGESIZE) |    // physical address
        PT_PAGE |     // it has the "Present" flag, which must be set, and we have area in it mapped by pages
        PT_AF |       // accessed flag. Without this we're going to have a Data Abort exception
        PT_USER |     // non-privileged
        PT_ISH |      // inner shareable
        PT_MEM;       // normal memory

That is wrong that is a level 2 block descriptor the only bits that should be set are bit 0 & 1 and bit 31 if it is Non Secure. 

You start non secured mode EL1 so the correct form is
paging[0]=(unsigned long)((unsigned char*)&_end+2*PAGESIZE) |    0x8000000000000003ul; // physical address

Those other bits you OR on fortunately are ignored as they only go as high as bit11 .. see bottom table here which is L1 description Figure D4-16
https://armv8-ref.codingbelief.com/en/chapter_d4/d43_1_vmsav8-64_translation_table_descriptor_formats.html

You do the same to the TTBR1, identity L1.

There is still more issues I am walking thru, do you want me to send you the fixes or do you want to wade thru it yourself.

bztsrc, 2018-06-17:
Thank you for your post!

About TTBR_ENABLE bit, my hardware does not work without it. I know it's undocumented, but if I don't set the least significant bit, I got an exception raised. I saw that in the original AArch64 arm-truster-firmware examples procided by ARM, and it worked for me. I don't set it in the paging tables, but only in the TTBR register.
https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c#L209
https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/lib/aarch64/arch.h#L458

Now about the other thing, I'm not sure what you're saying. Again, I've empirically tested that if you're not mapping 2M blocks, all paging table entries along the path must be set to page granule, otherwise a translation exception occurs. I've tested my code in both qemu and on real hardware, and the two are not always working according to the documentation. But I'll check my code again to be sure.

Cheers
bzt

closed 2018-06-17 by bztsrc

bztsrc, 2018-07-04:
I've checked.

First, setting the least significant bit in TTBR on ARM8.2 is mandatory, no matter what the doc says. https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c#L250

Second, I'm absolutely positive, that those bits in the table are not ignored, and changing them triggers exception (translation or data abort). You can't change page granule to block granule in L1 and L2 if you intend to use page granule in L3. The same stands with read/write access. Haven't checked EL0/EL1 access part, but I suppose it's also works the same way.

Again, I'm not using documentation, rather experiments, testing on both real hardware and qemu, as the doc known to have bugs.

Cheers,
bzt

Amit-Radur, 2018-11-07:
Hello @bztsrc , Tried testing your Virtual memory example with QEMU but you mentioned following in 
Makefile

qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial stdio

But there is no such emulation exist for raspi3, do you have some in-house rpi3 emulation somewhere?

marcinszajner, 2018-12-16:
Hi,

There is some changes in qemu? because it work without TTBR_ENABLE. Additionally it work on raspberry pi 3 A+. Arm cortex-A53 is ARMv8.0 revision, and in trust firmware they say about ARMv8.2 revision:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c#L267

This is a bit strange for me, but maybe it's could be just another bug in documentation.

bztsrc, 2018-12-17:
Hi,

Yes it seems there's another bug in the documentation. I'm willing to trust the trusted firmware code :-)
```
	/*
	 * Enable CnP bit so as to share page tables with all PEs. This
	 * is mandatory for ARMv8.2 implementations.
	 */
        ttbr0 |= TTBR_CNP_BIT;
```
Anyway, if you set it, it cause no trouble in qemu or on machines that works without CnP. But there are hardware which definitely require that bit, so it's better to have it set all the time. 

Cheers,
bzt

marcinszajner, 2018-12-21:
Hi again,

I have question about TTBR1, kernel L1. Why we adress paging with [512+511]? Why value to next level  need be on the end of page? TTBR0 look for me ok, but this TTBR1...  I looked some info about it few days but I didn't find anything...

best regards,
Lazor

bztsrc, 2018-12-23:
Hi Layor,

Since we have set up 4k pages, we have a tree that points to 4k tables, each containing 512 long integer pointers. Because that would be a pain in the ass to access in C, I rather did a little trick here: put all translation tables in a contiguous memory area, and indexed them with a single pages[] array in C. Therefore indexes 0-511 belong to the first page (which starts at &_end), and index 512-1023 belong to the second page (starting at &_end+PAGESIZE) etc. Hope it's clear. Maybe I should explain this a bit more in the readme.

About ttbr0/1: nope, that's correct. A positive address has zeros in it's most significant bits, therefore ttbr0 is used. For a negative address, most significant bits are all ones, so ttbr1 is used. Now because we don't use addresses in the middle, only at the begining of the memory and the top of it, the first bits after the most significant group are also either all zeros or all ones (and nothing else used in between). That means we will use either the 0th or the 511th index on a page, and not the entries in between. Is this clear so far?

Now if you set the same page in both ttbr0 and ttbr1, then yes, it would work, because you'll still have the proper mappings, but you'll also have two shadow areas. One, MMIO (which supposed to be mapped at -2M only) would be also mapped at the top of the positive address range (from 510M to 512M), and the identity mapped RAM would be also mapped in the negative address range (from -512M to -X, where -X<-2, I don't remember right now exactly how much memory we are mapping). Does this make sense to you?

We want:
```
ttbr0 +----------+
      | identity |
      |          |
      |          |
ttrb1 +----------+
      |          |
      |          |
      | MMIO     |
      +----------+
```
And with ttbr0==ttbr1 we would get:
```
ttbr0 +----------+
      | identity |  <- still accessible at correct address
      |          |
      | MMIO     |
ttrb1 +----------+
      | identity |
      |          |
      | MMIO     |  <- still accessible at correct address
      +----------+
```

Cheers,
bzt

marcinszajner, 2018-12-30:
Hi Zoltan,

Thanks for answer, now I understand :)
I find intresting document about armv8 and mmu:
https://static.docs.arm.com/100940/0100/armv8_a_address%20translation_100940_0100_en.pdf

If you add this pdf or link to your (virtual memory) lesson it could be useful for people.

Best regards,
Lazor

fabiopolimeni, 2021-11-03:
Hi @bztsrc,

So, I have been trying enabling the MMU for TTBR1 on a kernel I am writing for educational purposes for a couple of weeks now. Some of it is inspired by your examples. At the moment I am using both TTBR0 and TTBR1, which point to the same initial table (L1, 4 entries), and have an identity mapping setup.

The issue I am having occurs when I try NOT to use the TTBR0, as I would like to keep this available for a later stage when I will be able to run user space programs. https://forums.raspberrypi.com/viewtopic.php?p=1932744#p1932744

At some point I stumbled upon this discussion and I saw this

```
ttbr0 +----------+
      | identity |
      |          |
      |          |
ttrb1 +----------+
      |          |
      |          |
      | MMIO     |
      +----------+
```
I am wondering why such a choice? I mean, is it even possible to boot with the following mapping at all?

```
ttrb1 +----------+    we dont want these mappings
      | identity |  </
      |          |
      | MMIO     |  <- still accessible at correct address
      +----------+
```

I am asking just because you say _we don't want this_. Is this an unwanted situation because of your example or it is something we shouldn't/couldn't do?

Thanks,
Fabio

SniperYa-Gi, 2023-08-09:
> 

Who ever gave this a thumbs down was completely retarded.