bztsrc/raspi3-tutorial #33: Black screen on 09_framebuffer framebuffer tutorial

cirosantilli, 2018-06-19:
- e176e5e780200ba4980e2436f81ad64df57d58a6
- use `apt-get` `aarch64-linux-gnu-` as in https://github.com/bztsrc/raspi3-tutorial/issues/30 which on my Ubuntu 18.04 is GCC 7.3.0.
- QEMU v2.12.0 as in https://github.com/bztsrc/raspi3-tutorial/pull/29

Then:

```
cd 09_framebuffer
make
qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial stdio
```

SDL window opens, and the default resolution does change to become larger, but remains black.

I don't see any splash for other tutorials as mentioned on README.

bztsrc, 2018-06-21:
The rainbow splash is only shown on real hardware, not on qemu. I've checked the pre-compiled kernel8.img, it shows Homer in the middle of the screen as expected, on real hardware as well as in qemu. If you can't see anything, I suppose there must be something wrong with your qemu compilation or with your cross-compiler.
I strongly urge you to get a real hardware, as the tutorials are supposed to be run on that, and not in qemu. It's just a nice feature to have, but qemu is definitely not the main target.

closed 2018-06-21 by bztsrc

cirosantilli, 2018-06-21:
Hmmm, I ran it again as described on ticket and it worked. I must have been doing something wrong previously. Thanks.

cirosantilli, 2018-06-22:
OK, ran again on Ubuntu 18.04, and it fails, but works with `-vnc :1`, therefore some QEMU SDL bug. Interestingly though, this Linux kernel example does show the framebuffer: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/741f5215e9515c0d7179671f49fe1781f94e70e3#graphic-mode-arm

cirosantilli, 2018-06-22:
Bug report at: https://bugs.launchpad.net/qemu/+bug/1778182

bztsrc, 2018-06-22:
Ah I see, thanks for the info, hopefully helps others too.

I do not use Bugbuntu at all. For me it is working without `-vnc :1`, and the SDL window shows the RPi's framebuffer. Also I have compiled my qemu, I'm not using the stock binary.

cirosantilli, 2018-06-22:
@bztsrc which distro are you using? I compiled QEMU v2.12 from source too.

bztsrc, 2018-06-30:
I've finally had the time to compile the latest 2.12.50. It works perfectly for me, no need for vnc. I've used gcc 7.3.1, binutils 2.29.1 and sdl 2.0.8. Configured qemu with the following arguments:
`./configure --target-list=aarch64-softmmu --enable-modules --enable-tcg-interpreter --enable-debug-tcg --python=/usr/bin/python2.7`
and started the framebuffer tutorial with `make run`.