bztsrc/raspi3-tutorial #36: Will your raspbootin64 run 32-bit code?

profilernz, 2018-07-02:
I need a version of raspbootin that will run 32-bit ARMv7 code on the Pi3.
I suspect this is not possible?
I got your raspbootin64 booting and talking to raspbootcom, but it just got stuck in a loop of sending and resending my image.
I'm working on a project with lots of 32-bit assembly, so I can't easily compile my stuff to 64-bit...
My understanding is that the original raspbootin won't run on the Pi3... Any other suggestions?
Thanks!

bztsrc, 2018-07-03:
No, 32 bit code won't run in 64 bit mode. And I'm not planning to backport it, I've gave up programming for 32 bit more than a decade ago.

But you can backport easily though, here are my thoughts on it: there's only a minimal assembly in start.S, other than that everything is in C, should compile for 32 bit out-of-the-box and have a kernel7.img. (I'm not sure as I haven't tried 32 bit on RPi3 but maybe you have to change the base address of MMIO. RPi had it at different address, and I don't know whether that mapping is BCM specific or CPU mode specific. I'd guess it's BCM specific, so no need to change).

Also it is possible to load a 32 bit kernel with a 64 bit loader: alternatively you can copy the EL switch from one of the other tutorials (to where you hand over the control to the kernel), without setting the AArch64 flag. That way the loader would run in 64 bit mode, but the kernel would be executed in 32 bit mode in EL1 as you transfer control to it with an `eret`. In this case you'd have to relocate your loader code to 0x8000-1024 as 32 bit kernels are expected to be loaded at 0x8000 and not 0x80000. I think that's all, everything else should be fine. (Assuming you don't care your kernel is started in EL1 and not EL2. Linux switches to EL1 as soon as possible, so shouldn't be a problem)

Btw I don't see any reason why the original raspbootin shouldn't run on the RPi3 as kernel7.img, if the MMIO is BCM specific, you only have to change the MMIO base address to the one I use in raspbootin64 and good to go.

Good luck
bzt

bztsrc, 2018-07-03:
I take it back, it seems that the original raspbootin autodetects the MMIO address, so no change needed at all.
https://github.com/mrvn/raspbootin/blob/master/raspbootin/main.cc#L39
You should be able to boot it as kernel7.img on a RPi3.

bzt

closed 2018-07-03 by bztsrc

profilernz, 2018-07-06:
Thanks for your help and suggestions.
I just get garbled tty output booting as kernel7.img on a Pi3...

bztsrc, 2018-07-06:
Oh I see. That's definitely means incorrect baud rate, and taking a look at the original raspbootin's source I've found the problem. In
https://github.com/mrvn/raspbootin/blob/master/raspbootin/uart.cc#L80
it sets up the divisor for the baud rate  (IBRD=1,FBRD=40), but incorrectly assumes that UART_CLOCK = 3000000. That's not necessairly the case (at least not on RPi3).

To fix it, you should add a mailbox call in UART::init() at line 84 to anchor the UART clock at a known frequency. Here's an example how to do that (I've used 4MHz in my tutorial, but to be compatible with the original raspbootin, either you should use 3MHz, or keep 4MHz and change the divisor value to IBRD=1,FBRD=11):
https://github.com/bztsrc/raspi3-tutorial/blob/master/05_uart0/uart.c#L49

The mailbox interface is quite straigthforward, you need a 16 bytes aligned uint array and one function to pass the array's reference to the GPU (see mbox.c), once you've filled the array up with values (see uart.c).
https://github.com/bztsrc/raspi3-tutorial/blob/master/05_uart0/mbox.c

Once the UART clock is at a known freq, the baud rate should be consistent. With a little luck, this is the last and only obsticle you have to come around.

bzt

bztsrc, 2018-07-06:
A few more hints:
1. your mbox array needs to hold no more than 8 ints
2. because we are not interested in the return value, all you need from the mbox_call is line 49 and line 51, which resemblances the UART::putc() very very much.

That's all. So something like
```c
volatile unsigned int __attribute__((aligned(16))) mbox[8];

// in uart.cc at line 84
mbox[0]=8*4;
mbox[1]=0;
mbox[2]=0x38002;
mbox[3]=12;
mbox[4]=8;
mbox[5]=2;
mbox[6]=3000000;
mbox[7]=0;
while (true) {
  if (MMIO::read(0x0000B880 + 0x18) & 0x80000000) {
    break;
  }
}
MMIO::write(0x0000B880 + 0x20, &mbox | 8);
```
should do the trick. (Note I haven't tested this code, just put it together to give you the idea on a minimalistic modification).

bzt

bztsrc, 2020-03-16:
Hi,

It's been a while, but if you're still interested in this, I've finally implemented a multi-platform version of the boot loader which supports AArch32 (on RPi3 and RPi4 too).
https://gitlab.com/bztsrc/imgrecv
As for the image sender, see
https://gitlab.com/bztsrc/usbimager (you'll have to start it with the "-S" flag)

Cheers,
bzt

profilernz, 2020-03-20:
Wow, great!
Thanks for the update :)