bztsrc/raspi3-tutorial #41: File malloc

OllieLollie1, 2018-07-12:
Sorry for another malloc question but on line 190 of your fat.c in the readfile tutorial you use s of a way of knowing where to place your file data. How big can s get is there a way of knowing how much to allocate before running the sd_readblock command? Once that is done I guess I can use the dir->size as a way for knowing how much to malloc for the file.

bztsrc, 2018-07-12:
Well, bpb->nr holds the number of maximum root directory entries. Multiply it by sizeof(dir_t) and you'll get how much memory you need to read it entirely into memory. Also, data sectors (sectors holding subdirectory entries and file contents) are placed right after the root directory entries, so the same size is needed to locate file data on disk.

closed 2018-07-12 by bztsrc

OllieLollie1, 2018-07-12:
How big would that malloc get because as I look at it it would exceed the physical memory the pi has. The max a fat32 system can hold for one file ~4GB so wouldn't the malloc be around that size?

OllieLollie1, 2018-07-14:
Ok, I don't quite get what you mean. On line 130 of fat.c 
`if(sd_readblock(root_sec,(unsigned` char*)dir,s/512+1))`
Does that mean that I allocate for s/512+1?
On line 190
`s=sd_readblock(partitionlba+1,(unsigned char*)&_end+512,(bpb->spf16?bpb->spf16:bpb->spf32)+bpb->rsc);`
Does that mean that I should allocate (bpb->spf16?bpb->spf16:bpb->spf32)+bpb->rsc)?
On line 196
`sd_readblock((cluster-2)*bpb->spc+data_sec,ptr,bpb->spc);`
Does that mean that I should allocate bpb->spc?

If the file is > 10MB I would use the buffered read. Where when a particular amount is asked for provided it is < 10MB then that amount would be placed in the buffer. That method would be fairly simple as far as memory allocation goes however I'm feeling that efficient memory allocation is more difficult.

OllieLollie1, 2018-07-19:
Sorry but I still don't understand. Would you mind providing some pseudo code?

bztsrc, 2018-07-20:
Just rearrange the code.
in fat_getpartition, before return ok:
```c
// globals
fatdir_t *rootdir;
unsigned int *fat32;
unsigned int file_size=0;
unsigned int cluster_size=0;
unsinged in cluster=0;
  ...
fat_getpartition:
  // allocate buffers
  rootdir=malloc(x);
  fat32=malloc(x);
  // read root directory once
  root_sec=...
  sd_readblock(root_sec,rootdir,...
  // read FAT once
  data_sec=...
  sd_readblock(partitionlba+1,fat32,...
  // allocate content buffer
  cluster_size=bpb->spc*bpb->bps;
  buf=malloc(cluster_size);
  bufsize=cluster_size;
  return 1;
  ...
```
Now fat_getcluster would be:
```c
  dir=rootdir;
  for(;dir->name[0]!=0;dir++) {
     ...
     file_size=dir->size;
     cluster=((unsigned int)dir->ch)<<16|dir->cl;
     ...
```
And fat_readfile:
```c
   if(currptr>=bufsize || bufsize==-1) {
      // don't read end of chain cluster markers
      // also if last read was less than a full cluster, we're already at the end
      if(cluster<1 || cluster<0xFFF8 || bufsize!=cluster_size)
         return EOF;
      // read one cluster into buffer
      sd_readblock((cluster-2)*bpb->spc+data_sec,buf,bpb->spc);
      // get next cluster in advance
      cluster=bpb->spf16>0?fat16[cluster]:fat32[cluster];
      // calculate buffer size and remaining file size.
      // bufsize is always cluster size except for the last iteration, where it can be less
     if(file_size>cluster_size) {
       file_size -= cluster_size;
     } else {
       bufsize = file_size;
     }
      currptr=0;
   }
   if(bufsize==0)
     return EOF;
   return buf[currptr++];
```
Note, I haven't tested this code, but you can get the idea.
fat_getpartiton loads, allocates and calculates everthing in advance.
fat_getcluster uses the preloaded directory list to get the starting cluster and the file size.
fat_readfile returns the next character from the buffer. If buffer is empty or the pointer is at the end, we load the next cluster into the buffer, and return the first character in the buffer. Buffer size is always cluster_size, except for the last part, which can be 0..(cluster_size-1). Therefore we substract cluster_size from file_size each time we read a cluster, and at the end we use the remaining for the last bufsize.

Cheers,
bzt

OllieLollie1, 2018-07-23:
In your first chunk, how big is x? Is each allocation the same size or are they different?

Thanks,
OllieLollie_1

bztsrc, 2018-07-24:
I've already told you that in my second post. Use the fields in bpb to calculate the sizes. You have bpb->nr which holds the number of root directory entries, and you also know the size of one directory entry (32 bytes). Because you'll read 512 bytes at once, you should round that up to avoid buffer overflow.  Likewise, bpb->spf holds how many sectors required per FAT (there are two, but you only need to read one), and you also know the size of one sector (512 bytes).

Alltogether you'll need three allocations:
- root directory entries, nr*32, round it up to be multiple of 512 (read once)
- one copy of the FAT, spf*512 (read once)
- a cluster sized buffer, spc*512 (read data into many times)

You allocate only at the beginning, no need to allocate repeatedly later. You'll need root directory and FAT for every file, so don't free and clear those buffers ever (unless you change the media, which is not the case on RPi). And reading any file will re-use the same cluster buffer (regardless how long the file is, and how many files you read). With multiple open files, your cluster buffer array should have several cluster buffers, one for each open file, all of the same size, alloc'd on fat_getcluster and freed with fat_close. Even then you shouldn't free the root directory and FAT buffers.

Cheers,
bzt