bztsrc/raspi3-tutorial #45: 0B_readsector timeout waiting for ready to read

heidecjj, 2018-10-26:
I can get 0B_readsector to work in qemu, but I cannot get it to work on a Pi 3 B or a Pi 3 B+. I've tried using the prebuilt kernel8.img as well as building my own kernel8.img. The only files on my SD card are bootcode.bin, start.elf, fixup.dat, and kernel8.img. When the Pi boots, it successfully initializes the SD card, but times out when trying to read from the SD card with the following message:
`ERROR: Timeout waiting for ready to read`

heidecjj, 2018-10-29:
I think something was wrong with my SD card. It was a Class 10 SDHC but switching to another Class 10 card fixed my problem. Thanks!

closed 2018-10-29 by heidecjj

bztsrc, 2018-10-29:
You are welcome, glad to hear it's working now!

sjtugjy, 2020-10-29:
So, this tutorial can only work on Class 10 instead of SDHC?

Thanks for considering this. @bztsrc @heidecjj