bztsrc/raspi3-tutorial #47: some error when building up toolchains

eunwoo, 2018-11-02:
Hello,
I'm studying your tutorial. Thank you^^.
There's some error in building binutils.
Could you give me some help?
I followed your tutorial step by step.
I'm testing them by running Ubuntu 18.04.1 LTS in Virtual Box.

rpi@rpi-VirtualBox:~/binutils-2.30$ make -j4
make[1]: leaving directory '/home/rpi/binutils-2.30'
make[1]: nothing to do for 'all-target'
Configuring in ./etc
Configuring in ./intl
Configuring in ./isl
Configuring in ./libiberty
configure: error: cannot find sources () in .././isl
Makefile:5750: recipe for target 'configure-isl' failed
make[1]: *** [configure-isl] Error 1
make[1]: *** waiting for unended jobs....
configure: creating cache ./config.cache
checking whether make sets $(MAKE)... configure: creating cache ./config.cache
checking for a BSD-compatible install... /usr/bin/install -c
configure: creating cache ./config.cache
checking whether to enable maintainer-specific portions of Makefiles... no
checking for makeinfo... /home/rpi/binutils-2.30/missing makeinfo --split-size=5000000
configure: WARNING:
*** Makeinfo is missing. Info documentation will not be built.
checking for perl... perl
yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether NLS is requested... yes
checking for msgfmt... checking build system type... configure: updating cache ./config.cache
/usr/bin/msgfmt
checking for gmsgfmt... /usr/bin/msgfmt
configure: creating ./config.status
checking for xgettext... x86_64-pc-linux-gnu
checking host system type... /usr/bin/xgettext
x86_64-pc-linux-gnu
checking for x86_64-pc-linux-gnu-ar... ar
checking for x86_64-pc-linux-gnu-ranlib... ranlib
checking whether to install libiberty headers and static library... yes
configure: target_header_dir = libiberty
checking for x86_64-pc-linux-gnu-gcc... gcc
checking for msgmerge... /usr/bin/msgmerge
checking for x86_64-pc-linux-gnu-gcc... gcc
checking for C compiler default output file name... checking for C compiler default output file name... a.out
checking whether the C compiler works... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... yes
checking whether we are cross compiling... no
checking for suffix of executables... config.status: creating Makefile

checking for suffix of object files... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... yes
checking for gcc option to accept ISO C89... none needed
none needed
checking how to run the C preprocessor... checking build system type... x86_64-pc-linux-gnu
checking host system type... gcc -E
x86_64-pc-linux-gnu
checking for x86_64-pc-linux-gnu-ranlib... ranlib
checking for library containing strerror... checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... none required
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
[... 286 lines cut ...]
Makefile:850: recipe for target 'all' failed
make: *** [all] Error 2
rpi@rpi-VirtualBox:~/binutils-2.30$ 

This is the some part related to isl in Makefile.
.PHONY: maybe-maintainer-clean-isl maintainer-clean-isl
maybe-maintainer-clean-isl:
maybe-maintainer-clean-isl: maintainer-clean-isl

maintainer-clean-isl: 
	@[ -f ./isl/Makefile ] || exit 0; \
	r=`${PWD_COMMAND}`; export r; \
	s=`cd $(srcdir); ${PWD_COMMAND}`; export s; \
	$(HOST_EXPORTS) \
	for flag in $(EXTRA_HOST_FLAGS) V=1; do \
	  eval `echo "$$flag" | sed -e "s|^\([^=]*\)=\(.*\)|\1='\2'; export \1|"`; \
	done; \
	echo "Doing maintainer-clean in isl"; \
	(cd $(HOST_SUBDIR)/isl && \
	  $(MAKE) $(BASE_FLAGS_TO_PASS) "AR=$${AR}" "AS=$${AS}" \
	          "CC=$${CC}" "CXX=$${CXX}" "LD=$${LD}" "NM=$${NM}" \
	          "RANLIB=$${RANLIB}" \
	          "DLLTOOL=$${DLLTOOL}" "WINDRES=$${WINDRES}" "WINDMC=$${WINDMC}" \
	          maintainer-clean) \
	  || exit 1

bztsrc, 2018-11-02:
Wrong forum. As you can read on the readme,
> IMPORTANT NOTE: this description is not about how to compile a cross-compiler in general. It's about how to compile one specifically for the aarch64-elf target. **If you have problems, google "how to build a gcc cross-compiler" or ask on an appropriate support forum for your operating system.** I can't and won't help you with your environment, you have to solve that on your own.

I suggest to ask your question on a gcc forum. But if I were you, I'd start by investigating the first error message in your ourput:

> configure: error: cannot find sources () in .././isl
> Makefile:5750: recipe for target 'configure-isl' failed

Cheers,
bzt

closed 2018-11-02 by bztsrc

adrianlopezroche, 2018-11-05:
Hi @bztsrc,

I think this is actually a bug in your instructions. The bug has to do with the link to `isl-0.18` under `binutils-2.30` (I'm using the same versions as in your tutorial). Specifically:

```
cd binutils-*
ln -s ../isl-* isl
```

will create a symlink named `isl` under `binutils-2.30`, but `isl` is not a real directory. Because it isn't a real directory, `isl/..` will point not to the parent of the symlink `isl` (namely, `binutils-2.30`) as might be expected, but to the parent of the directory `isl-0.18`. Thus, for a directory structure such as this one:

```
/path/to/sources/isl-0.18
/path/to/sources/binutils-2.30
```

`binutils-2.30/isl/..` will point to `/path/to/sources` rather than `binutils-2.30`. This is significant because the `Makefile` produced by `./configure` calls `cd ./isl` and expects `.././isl` to point to `isl` under `/path/to/sources/binutils-2.30` which it doesn't: it points to `/path/to/sources/./isl` instead, which doesn't exist and breaks the `Makefile` script. with the following error:
```
configure: error: cannot find sources () in .././isl
```

You can reproduce the effect with a simple experiment:

```
mkdir parent
cd parent
mkdir child1
mkdir child2
cd child2
ln -s ../child1 c1link
ls c1link/..
```
This will list the contents of `parent` rather than the contents of `child2`.

You've stated elsewhere that your example works on your system, but this is the reason it fails on mine and @eunwoo's. Are you using the exact versions of bintools and gcc listed in your tutorial?

bztsrc, 2018-11-05:
Hi,

Yes, I've used binutils-2.30 and isl-0.18. But I haven't used that readme tbh (I already had the cross-compiler when I wrote that), and also I've merged a patch from someone else since, so there could be some error, although I really doubt that (the patch was mostly about checksums).

Here's my full cross-compiler compiling shell script that I've actually used (and I wrote the original readme after this, removing parts that are irrelevant for the tutorials, and I've unrolled a few loops for clearity):
```sh
#!/bin/bash

### Configuration
ARCHS="x86_64 aarch64"
DIR="/usr/local/gcc"
CFG=" --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit \
--disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-plugin \
--disable-linker-build-id --enable-lto --enable-install-libiberty --with-linker-hash-style=gnu --with-gnu-ld\
--enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie \
--enable-default-ssp --enable-gnu-unique-object"

### Get binutils and gcc source. Modify versions in filenames as you like. Only place where versions referenced
[ ! -d tarballs ] && mkdir tarballs
echo "Downloading tarballs..."
cd tarballs
[ ! -f binutils-*tar* ] && wget http://ftpmirror.gnu.org/binutils/binutils-2.30.tar.gz
[ ! -f gcc-*tar* ] && wget http://ftpmirror.gnu.org/gcc/gcc-7.3.0/gcc-7.3.0.tar.gz
[ ! -f mpfr-*tar* ] && wget http://ftpmirror.gnu.org/mpfr/mpfr-4.0.1.tar.gz
[ ! -f gmp-*tar* ] && wget http://ftpmirror.gnu.org/gmp/gmp-6.1.2.tar.bz2
[ ! -f mpc-*tar* ] && wget http://ftpmirror.gnu.org/mpc/mpc-1.1.0.tar.gz
[ ! -f isl-*tar* ] && wget ftp://gcc.gnu.org/pub/gcc/infrastructure/isl-0.18.tar.bz2
[ ! -f cloog-*tar* ] && wget ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.1.tar.gz
cd ..

### Don't change anything below this line
echo -n "Unpacking tarballs... "
for i in tarballs/*.tar.gz; do d=${i%%.tar*};d=${d#*/}; echo -n "$d "; [ ! -d $d ] && tar -xzf $i; done
for i in tarballs/*.tar.bz2; do d=${i%%.tar*};d=${d#*/}; echo -n "$d "; [ ! -d $d ] && tar -xjf $i; done
echo "OK"

cd binutils-* && for i in isl; do ln -s ../$i-* $i 2>/dev/null || true; done && cd ..
cd gcc-* && for i in isl mpfr gmp mpc cloog; do ln -s ../$i-* $i 2>/dev/null || true; done && cd ..

#patch buggy gcc configure script to enable -fvisibility in gcc
cd gcc-*/gcc
cat configure|sed 's/gcc_cv_as_hidden=no/gcc_cv_as_hidden=yes/'|sed 's/gcc_cv_ld_hidden=no/gcc_cv_ld_hidden=yes/'>..c
mv ..c configure
chmod +x configure
cd ../..

### Compile gcc
for arch in $ARCHS; do
    echo "  -------------------------------------------- $arch -----------------------------------------------------"
    mkdir $arch-binutils 2>/dev/null
    cd $arch-binutils
    ../binutils-*/configure --prefix=$DIR --target=${arch}-elf $CFG
    make -j4
    make install
    cd ..
    mkdir $arch-gcc 2>/dev/null
    cd $arch-gcc
    ../gcc-*/configure --prefix=$DIR --target=${arch}-elf --enable-languages=c $CFG
    make -j4 all-gcc
    make install-gcc
    cd ..
done
```
I can still run this script without errors. I think the only difference is the gcc version, as it uses 7.3.0. Try my script, maybe that helps. Ubuntu by the way was reported to have broken gcc by several users, and some users have reported issues with compiling the sources too (interestingly also with problems related to isl). I'd suggest to use a different distro if you can, or ask around on Ubuntu forums. As a last resort, remove the symlink and copy the contents of isl-0.18 into binutils/isl and gcc/isl.

Good luck,
bzt

adrianlopezroche, 2018-11-06:
Thank you. Looking at your script, I think I've spotted the reason why it works for you when it fails for us: you're creating a new directory for each target and compiling from there. The `configure-isl` rule for the `Makefile` generated by the bintools `configure` script will _not_ work when compiling directly inside `bintools-2.30`, but it works OK when compiling from a new directory (I won't bother going into why).

I haven't run your full script, but bintools-2.30 compiles successfully on Linux Mint (an Ubuntu-based distro) if I alter the steps of your tutorial (under "compiling the sources") as follows:

```
mkdir aarch64-binutils
cd aarch64-binutils
../binutils-*/configure --prefix=/usr/local/cross-compiler --target=aarch64-elf \
--enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit \
--disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-plugin \
--disable-linker-build-id --enable-lto --enable-install-libiberty --with-linker-hash-style=gnu --with-gnu-ld\
--enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie \
--enable-default-ssp --enable-gnu-unique-object
make -j4
make install
```

I haven't tried compiling gcc yet, but a similar edit is probably required.

bztsrc, 2018-11-06:
> The configure-isl rule for the Makefile generated by the bintools configure script will not work when compiling directly inside bintools-2.30, but it works OK when compiling from a new directory

Wow, I've never thought of that not even in my wildest dreams! After all the '../isl-0.18' symlink in both binutils-2.30/isl and aarch64-binutils/isl should point to the same directory. This isl issue was puzzling for quite some time now, so thank you for spotting that!

> I won't bother going into why

I can understand that :-)

I've updated the readme, and since my gcc compilation also uses a new directory (which works for me) I've added that too.

Big thanks,
bzt