bztsrc/raspi3-tutorial #48: 03_uart1 - bug

marcinszajner, 2018-11-04:
Hi,
I doing your tutorial and because I don't know yet this architecture I compare most of thinks with documentation.
I find out mistake in 03_uart1 - uart.c:56 *AUX_MU_IIR = 0xc6;    // disable interrupts
I think it should be 0xc0
because page 13:
https://web.stanford.edu/class/cs140e/docs/BCM2837-ARM-Peripherals.pdf

and another minor
 uart.c:55     *AUX_MU_IER = 0;
dublicated line

Best regards,
Marcin

bztsrc, 2018-11-04:
Hey, thank you!
Indeed, that's a duplicate line! I'll check the 0xc6 issue too, because documentation not always up-to-date, and sometimes real hardware works differently. I'll look into that! Thanks again for noticing!

bzt

bztsrc, 2018-11-05:
Ok, checked, it turned out everything is correct.

About 0xC6 in IIR, 6 stands for 2+4, so bits 1 and 2. Accoring to the docs, writing those bits in IIR should clear both the receiver and transmitter FIFO queues. So if you use 0xC0, your code will probably work most of the time (provided you've just turned on your Pi), but could act sometimes "strange" if certain conditions met (when there are leftover bytes in one of the queues, which could happen after a reset through a PM call). The best practice is to clear data registers (FIFO queues in this case) during initialization regardless their state, just to be on the safe side.

Also those IER lines really look like duplicates, but actually they are not. :-) The beauty of low level programming. The first clears the baud rate, the second disables the FIFO interrupts. We change the MMIO / AUX register mapping in line 53 by clearing DLAB bit in LCR when we set 8 bit data mode.

Cheers,
bzt

closed 2018-11-05 by bztsrc

marcinszajner, 2018-11-05:
Awww I incorrectly read datasheet. Now I see, it was about read not a write condition.

Hmm DLAB bit in LCR shouldn't be reset state if we not set it somewhere before? If DLAB is set somewhere and we write 0 to IER register that mean we only clear MS 8bits from 16bit baudrate register isn't it?
Low level programing is as good as SoC documentation is good  :D

bztsrc, 2018-11-05:
Yes you are right about the DLAB bit. The code only clears MS 8bits after a turn on, after a PM reset DLAB should be 0, and I haven't set it explicitly before the first line. So it is a duplicated line after all :-) We can remove the first one probably, but I'll give it a test just to be sure. (Thankfully it doesn't matter because the code sets the baud rate register by using a different MMIO address too.)

Cheers,
bzt

marcinszajner, 2018-11-05:
If I find some other bug or something that look for me strange, do you want me to report it?

BTW, nice tutorial. It is really helped me a lot :)

bztsrc, 2018-11-06:
Yes, but of course! Thank you very much, and I'm glad my tutorials were useful! :-)

Cheers,
bzt