bztsrc/raspi3-tutorial #49: uart0 - incorect mailbox

marcinszajner, 2018-11-06:
Hi again,

I find something new, that not look for me ok, but code work well on qemu.
In 05_uart0/uart.c
When you set message line 54 mbox[4] = 8;
I think it should be 0; Why? In mailbox property interface:
https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface
we have tag format:
u32: tag identifier
u32: value buffer size in bytes
u32:
    Request codes:
        b31 clear: request
        b30-b0: reserved
    Response codes:
        b31 set: response
        b30-b0: value length in bytes
u8...: value buffer

In your code it is done in this way:
    mbox[2] = MBOX_TAG_SETCLKRATE; // set clock rate
    mbox[3] = 12;
    mbox[4] = 8;
    mbox[5] = 2;           // UART clock
    mbox[6] = 4000000; // 4Mhz

after tag identifier and buffer size should be 4 bytes with code (u32: without label), that should be 0 (because b30-b0 is reserved).

Additional tag 0x00038002 (Set clock rate) have 3 input argument (clock id, rate and skip setting turbo)

In your code you put correct size  (12) but incorect number of arguments (clock id and rate). That mean mbox[0] should be 9*4 and between mbox[6] and mbox[7] should be additional element with skip setting turbo argument (0 or 1).
Of course in qemu we cannot set clock rate and I didn't check it on target, however from "mailbox property interface" it look incorect a bit (but propably still work).

Best regards,
Marcin

bztsrc, 2018-11-08:
Hi,

Thanks for the through testing!

The mbox[4]=8 is correct, because "set clock rate" has an 8 bytes response, see https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#set-clock-rate
According to my tests, with some tags on real hardware you have to tell the GPU how much space you've allocated for the response, so I like to set the response buffer size for all tags (won't hurt even if it's not strictly required). You'll see that with the LFB Mailbox as well. Spending some time with ARM documentations and you'll realize that they are sometimes contradictional, and real hardware does not necessarily work as described.

And about the 12 bytes input, you're right. I've copied this code from my production ready bootloader, where readibility wasn't a goal. It's working because the "last tag" macro is 0, so this also works as a third parameter which clears turbo as intended. To be 100% correct, it should be
```
mbox[0]=9*4;
...
mbox[7]=0;
mbox[8]=MBOX_TAG_LAST;
```
But doesn't really matter because now mbox[7] is zero anyway and we have mbox[0]=8*4, so GPU won't read mbox[8] (MBOX_TAG_LAST is not mandatory, mbox size is more important).

But I'll fix that to make the code clearer, give me a day or so, unfortunately right now I've other things to attend to. If you find anything else in the meantime, I'll commit them too.

Thanks again,
bzt

marcinszajner, 2018-11-08:
hmm compare to this tutorial:
http://www.valvers.com/open-software/raspberry-pi/step05-bare-metal-programming-in-c-pt5/

After set tag and size of buffer is olways 0 as /* Request */. I didn't find anywhere info about setting response buffer size.

bztsrc, 2018-11-10:
> hmm compare to this tutorial:

That tutorial has other issues as well, although I give you that using va args for mbox tags is pretty smart and elegant. Btw that's not for BCM2837, and I'm not sure they have tested it on RPi3 at all.

> I didn't find anywhere info about setting response buffer size.

Me neither :-) I know that empirically. I've also made that mistake first to trust ARM documentation. Never do that. Use any documentation as a guide, and ALWAYS test your code on a real machine. I've spent more than a week figuring out what's wrong with my MMU code. As it turned out, it was a documentation issue, as the official ARM Developer Guide neglected to mention that the encoding of 4k page size in TCR *differs* for TTBR0 and TTBR1. Who'd have thought of that?
Also compare the doc with the arm trusted firmware example code. There'll be lots of differences, but thankfully you can try out the code :-) For example [this little nuace](https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c#L267) about CnP bit is not mentioned in the docs at all (at least it wasn't last time I checked).
Oh, and always look for errata. There lots of fixes for the BCM2835/6/7, which can save your day.

bzt

closed 2018-11-10 by bztsrc

toothbrush7777777, 2019-09-23:
I know that this is a little late, but I'd like to point out that the mailbox interface is not hardware and is monitored by the software running on the GPU.