bztsrc/raspi3-tutorial #51: sd.c

OllieLollie1, 2018-12-31:
When you in your code have readblock you say you read a block from the sd card and return the number of bytes read. What is a block in this case? I have written a fat driver but the current interface uses a function that takes (uint32_t sector, uint8_t *buffer, uint32_t sector_count) but I'm not sure how your interface works so I can't yet adapt it. Would you please tell me what you have done?

bztsrc, 2018-12-31:
Hi,

The block size is 512 bytes, so the readblock function returns multiples of 512 bytes.
That readblock should be easy to intergrate with your fat driver. It works like this: you give the starting lba of the sector to read, a buffer in memory to read to, and finally the number of blocks (sectors) to read. When it returns count*512, then all of the sector contents are read into the buffer. Just a sidenote, it should return 0 on error or count*512 on success. The only case when it returns a non-zero, but less value is when you read from the end of the medium.
The reason it expects sector address and count but returns bytes is simple: when you calculate addresses for fat, you usually got the input in sectors. And it returns bytes because you most likely want to adjust the buffer address with that.

Cheers,
bzt

closed 2018-12-31 by bztsrc

bztsrc, 2019-01-12:
Hi,

Yes, lba is the sector number.

Cheers,
bzt