bztsrc/raspi3-tutorial #52: multicore - incorrect?

marcinszajner, 2019-02-24:
Hi again,

I start analyse another tutorials like Leon de Boer's tutorials. Your code look like all core start at the same entry point that why you check there ID and stop it. in LdB toturial in smartStart64.S;

https://github.com/bztsrc/raspi3-tutorial/blob/master/02_multicorec/start.S#L31

bztsrc, 2019-02-25:
Hi,

Take a closer look at LdB's code, he's code is executed on all cores too, and he is stopping cores too.
Line 57: there he reads the core's id in which the code is running
Line 81: there he reads core id again to set up different stacks per core
Line 147: finally, there he reads core id once more, and core0 jumps to cpu0_exit_multicore_park while core 1,2,3 continues to StartSecondarySpin

Whether the kernel.img is executed on all cores at once or only on core0, depends on config.txt (kernel_old), so it's better to be prepared for any configuration. That's why my code also has an EL3 to EL2 switch too.

Cheers,
bzt

closed 2019-02-25 by bztsrc

marcinszajner, 2019-02-25:
https://www.raspberrypi.org/forums/viewtopic.php?t=232895

This discusion cleared  my doubts. Thanks for answer :)

Best regards,
Lazor