bztsrc/raspi3-tutorial #55: handling big memory

jsmith173, 2019-06-25:
Do I need any special code to run this code? In general what I need to handle 'big memory'?

#include <stdlib.h>

int main(void)
{
 int size = 500*1024*1024;
 int pos = 200*1024*1024;
 unsigned char* p = malloc(size);
 p[pos] = 0xAA;
 unsigned char c = p[pos];

 while (1) ;
}

Ace17, 2019-06-25:
Have you tried enabling compiler warnings? ( -Wall -Wextra )

jsmith173, 2019-06-25:
> Have you tried enabling compiler warnings? ( -Wall -Wextra )

Sorry, * marks somehow removed

void kernel_main( unsigned int r0, unsigned int r1, unsigned int atags )
{
    int size = 500*1024*1024;
    int pos = 200*1024*1024;
    unsigned char* p = malloc(size);
    p[pos] = 0xAA;
    unsigned char c = p[pos];

    while (1) ;
}

jsmith173, 2019-06-25:
I don't know why mult. marks removed

int size = 500 mult 1024 mult 1024; //500MB
int pos = 200 mult 10241 mult 024; //200MB

bztsrc, 2019-06-25:
Don't worry about the asterix, I can see what you meant.

Normally you don't have to care about big memory save checking if the allocation was successful.
```c
p = malloc(size);
if (p) {
  p[pos] = 0xAA;
  ...
```

But what I don't get, is that neither of these tutorials allocate memory, because they are for bare metal without libc, and I haven't provided any malloc() implementation for them. So in short I don't think this is the right place for your question, therefore I close this ticket.

Cheers,
bzt

closed 2019-06-25 by bztsrc

bztsrc, 2019-06-25:
Just in case, if you meant to ask what do you need for allocating dynamic memory on bare metal, then the answer is:
- check out gcc's __builtin_malloc
- compile a well known allocator into your bare metal app, like dlmalloc, ptmalloc (there's more, like jemalloc, tcmalloc, Hoard etc., but I wouldn't recommend them as they are very hard to port). Now there's mimalloc, which I haven't used yet, but also looks promising.
- in order for any memory allocator to work, you'll need a physical memory allocator, which is usually called by sbrk or mmap system calls. Because of this I think the easiest solution would be to integrate dlmalloc, and provide a simple sbrk implementation for it like
```c
int memtop = &_end; /* use your linker script to get the end of your app, or choose an address like 4M for example */
int sbrk(int diff)
{
    return memtop += diff;
}
```
Note this is a dirty hack, not a proper physical memory allocator implementation, but for a single bare metal application with a single thread it will work just fine.

Cheers,
bzt

bztsrc, 2019-06-25:
https://github.com/ARMmbed/dlmalloc/blob/master/source/dlmalloc.c
It needs mbed_sbrk(intptr_t increment) to be implemented, which can be as simple as I showed you above, just a pointer to the beginning of free physical memory.

toothbrush7777777, 2019-08-17:
@jsmith173
BTW, the reason your asterisks ‘disappeared’ is that [comments (among other content) on GitHub are parsed as Markdown](https://guides.github.com/features/mastering-markdown/). Use the following to enclose code:

    ```c
    int main(void)
    {
        while (1);
    }
    ```