bztsrc/raspi3-tutorial #57: 14_raspbootin64 does not work.

T-Duke, 2019-07-16:
Hello!

I'v compiled several examples by clang 8.0. Some examples work fine. But serial kernel loader from exmp14 - don't.

It seems that  first kernel do not start at all. I can't see even RBIN64 message on terminal, after powering up my RPI3. But example 12_printf, works just fine.

bztsrc, 2019-07-16:
If you can't see the RBIN64 message, then the kernel is not running for sure. This could be a set up issue on your side, but also something else. It seems the problem only appears with Clang compiled kernels. Let me test them with the latest Clang, and I'll come back to you.

Cheers,
bzt

T-Duke, 2019-07-16:
Not only clang compiled images don't wjrk, but also original ones from 10,11,13,14 examples.
I don't understand why... Same SD card, same board, but only those examples does not work including original images.

09_framebuffer works in original image. Compiled by clang seems to start, I can see that TFT controller try to do something (my RPI3 connected to original TFT touchscreen), but picture is absent, only dark screen.

T-Duke, 2019-07-17:
UPD

I,v tried some other bootcode.bin in root of sSD card, and original precompiled images start to work properly. Some images compiled by clang also starts to work but not correctly. 13_debugger  fo example starts, breakpoint is reached, help command works, but when I try to dump registers, pressing  r, then all hangs.

14_raspbootin64 original precompiled image works fine and clang version to.

It seem that problems in some, more complex examples (virtual memory, debugger) strongly related to clang.

bztsrc, 2019-07-17:
Hi

I was unable to reproduce your issue, and with the correct firmware files all examples work correctly for you too. The problem is now limited to the Clang compiled virtual memory and debugger examples. Is this correct?
I'll focus on these two. It is possible that Clang changed the default arch and therefore compiles instructions which are not supported on the RPi3 (like armv8a or neon instructions). I'll look into that, probably could be solved by adding -march to the command line or enabling FPU if that's the case. As dumping registers is nothing more just a print command, I guess data access may cause the hang too. Does dumping one register, like "r x0" work? Although both unknown instruction and data abort inside the debugger should print out an error, so it's strange at least.

Cheers,
bzt

T-Duke, 2019-07-17:
Here we are is some screen video about clang compiled debugger issue: https://youtu.be/pw7DbvZHdF4

All works fine except debugger commands. Help screen prints normally, continue command also works, but other cmd don't.

bztsrc, 2019-11-02:
Adding mcpu flag to Clang solved this for me. If it reappears, let me know.

Cheers,
bzt

closed 2019-11-02 by bztsrc