bztsrc/raspi3-tutorial #58: How does the .bss get cleared?

connorkuehl, 2019-10-05:
Does `x1` get incremented somehow? I think `x1` has the same value throughout the loop so it looks like with every iteration a zero is being written to the same place.

In start.S:

```
    // clear bss
    ldr     x1, =__bss_start
    ldr     w2, =__bss_size
3:  cbz     w2, 4f
    str     xzr, [x1], #8
    sub     w2, w2, #1
    cbnz    w2, 3b
```

I'm also curious about the immediate offset in the `str`. Is it necessary?

connorkuehl, 2019-10-05:
Actually, after stepping through it with `gdb` it would appear the `str` instruction can update the base register. See the "Index Modes" sub section in the [ARM Cortex-A Series Programmer's Guide for ARMv8](https://developer.arm.com/docs/den0024/latest/the-a64-instruction-set/memory-access-instructions/specifying-the-address-for-a-load-or-store-instruction)

closed 2019-10-05 by connorkuehl

bztsrc, 2019-10-09:
Hi,

That's right. The #8 constant in the str instruction tells the CPU to adjust the offset register by 8 bytes.

Cheers,
bzt