bztsrc/raspi3-tutorial #60: udisasm usage suggestion

KenDickey, 2019-11-13:
udisasm is really useful, but it prints addresses in hex and requires addresses in decimal.

The "i" command would be more helpful if it took addresses in hex.

Thanks a bunch for the great tutorial!!

bztsrc, 2019-11-15:
Hi,

Thank you very much for your feedback!

It already can took addresses in hex. All offset arguments (not just for "i", but also for "x") are read by the dbg_getoffs() function, which understands the following formats:
1. "1234" decimal absolute offset
2. "0x1234" hexadecimal absolute offset
3. "x1+1234" decimal register relative forward offset
4. "x1-1234" decimal register relative backward offset
5. "x1+0x1234" hexadecimal register relative forward offset
6. "x1-0x1234" hexadecimal register relative backward offset

Cheers,
bzt

closed 2019-11-15 by bztsrc

KenDickey, 2019-11-15:
On 2019-11-15 11:55, Zoltan Baldaszti wrote:
..
> It already can took addresses in hex.

Ah!  Silly me.  I had not thought to prefix.

I was using a hex calculator to translate addresses.

Thanks once again for sharing your wonderful work!
-KenD