bztsrc/raspi3-tutorial #61: Issue accessing pcscreenfont on hardware

nickverlinden, 2019-12-09:
Hi,

I've run against a bit of a mystery, maybe due to my lack of proper knowledge.

I built a small kernell that doesn't really do anything yet, but draw some shapes on the framebuffer. That works fine. Then I added 5 different psf fonts, just like in the tutorial. When running in qemu, I can use all 5 fonts, without any problem. When running that very same kernell on a real pi3b+, the code crashes when using two specific fonts. The crash occurs at the first time I want to access a property of the psf struct.

I've been trying various things, but am desperate. I have absolutely no clue what to do. I could convert the psf fonts to c header files, but would rather go with the linking solution. I've read somewhere about alignment issues, but my lack of knowledge on this stuff prevents me from properly searching for a solution.

Kind regards,
Nick

bztsrc, 2019-12-10:
Hi Nick,

It is very difficult to say what could be wrong. I'd definitely suggest to at least add the exception handler from the https://github.com/bztsrc/raspi3-tutorial/tree/master/11_exceptions tutorial. That way you'll see what the problem actually is.

I don't think it's an alignment issue. Check the ELF (before you convert it into kernel.img), use objdump or readelf to list the symbols. The psf label should be aligned by default (and all fields in the psf header struct are aligned as well, so should be no problem). But who knows? The best to print out the ESR register, then look up in the DDI0478 manual what the error code actually means (the exc.c in the tutorial decodes some common exception codes, but not all of them).

Oh, and an unrelated thing: make sure you use psf files that DOES NOT contain unicode table, because this simple tutorial can't decode that (flags in psf header struct should be zero.) To handle psfs with unicode table, then first you'll need memory allocation (you could allocate statically the translation table, but that would add 128K to your kernel), otherwise it's easy. I wrote about it here: https://wiki.osdev.org/PC_Screen_Font (sample C code included).

I'll close this issue since it's not related to the tutorial's code, but let me know how it goes, what is the ESR and if you have more questions, I'll try to answer.

Cheers,
bzt

closed 2019-12-10 by bztsrc

nickverlinden, 2019-12-12:
Hi bzt,

Thanks for your reply, I've ordered a usb serial cable from china, but I'll have to wait a bit before I receive it. I already implemented the exception handler, so that should spread some light on the issue. For now I'm using xdd to convert the fonts to include files, and that solves the problem for now.

I have another issue that is just chewing my brains out. I have been on this (presumably) silly thing an entire day now, and still can't find what the problem is. I searched all over google, but can't find anything related to the pi 3b+. Maybe you could give me a hint where to look next. 
Basically I'm trying to set up uart0 at the midi baud rate 31250. So I set the clock frequency to 3mhz, and set the divider to 6, which results in 31250. The problem is, the midi data that is coming in is incorrect (i print the midi output to the screen). So there must be something wrong.
`
/* set up clock for consistent divisor values */
    mbox[0] = 9*4;
    mbox[1] = MBOX_REQUEST;
    mbox[2] = MBOX_TAG_SETCLKRATE; // set clock rate
    mbox[3] = 12;
    mbox[4] = 8;
    //mbox[5] = 2;           // UART clock
    mbox[5] = 2;           // UART clock midi
    //mbox[6] = 48000000;
    //mbox[6] = 4000000;     // 4Mhz
    mbox[6] = 3000000;     // 3Mhz midi
    mbox[7] = 0;           // clear turbo
    mbox[8] = MBOX_TAG_LAST;
    mbox_call(MBOX_CH_PROP);

/* map UART0 to GPIO pins */
    r=*GPFSEL1;
    r&=~((7<<12)|(7<<15)); // gpio14, gpio15
    r|=(4<<12)|(4<<15);    // alt0
    *GPFSEL1 = r;
    *GPPUD = 0;            // enable pins 14 and 15
    r=150; while(r--) { asm volatile("nop"); }
    *GPPUDCLK0 = (1<<14)|(1<<15);
    r=150; while(r--) { asm volatile("nop"); }
    *GPPUDCLK0 = 0;        // flush GPIO setup

    *UART0_ICR = 0x7FF;    // clear interrupts
*UART0_IBRD = 6; //((int)mbox[6] / (16 * 31250));//6;
    *UART0_FBRD = 0;
    *UART0_LCRH = (1 << 4) | (1 << 5) | (1 << 6); // Enable FIFO & 8 bit data transmission (1 stop bit, no parity).
    *UART0_IMSC = (1 << 1) | (1 << 4) | (1 << 5) | (1 << 6) | (1 << 7) | (1 << 8) | (1 << 9) | (1 << 10); // Mask all interrupts.
    *UART0_CR = (1 << 0) | (1 << 8) | (1 << 9); // Enable UART0, receive & transfer part of UART.
`

Investigating this problem further, I tried to read the clock frequency using a mailbox:
`// get clock rate
    mbox[0] = 12*4;
    mbox[1] = MBOX_REQUEST;
    mbox[2] = 0x30002;
    mbox[3] = 4;
    mbox[4] = 8;
    mbox[5] = 2; // clock id, 2 = UART clock
    mbox[6] = 0;
    mbox[7] = 0;
    mbox[8] = MBOX_TAG_LAST;
    if (mbox_call(MBOX_CH_PROP)) {
        printf("uart clock [id = %d, mhz = %d]", mbox[7], mbox[6]);
    }
    else {
        printf("Unable to query uart clock rate!");
    }`
On QEMU this gives me 3000000, on real hardware, this gives me 0. (before and after setting the baud rate using the mailbox).

Do you perhaps have any idea what I'm doing wrong? I also tried setting the clock to 4mhz and the divider to 8, should be resulting in 31250. Doesn't seem to work either, the midi data still is not correct.

bztsrc, 2019-12-13:
About querying the current clock, the first mbox[0] is wrong, should be 7*4 (or 6*4), otherwise looks good (the id is supposed to be returned in mbox[5], but regardless Hz should be in mbox[6]). Try to pass 8 as mbox[3] maybe?

About the clock rate, yeah, it should be ok, but these crystals in the UART chip are tricky things, not sure if it can resonate at multiple of 31250 Hz. I've only used 19200, 38400 bauds. You see, the original 8250 chip in the IBM PC (that RPi tries to mimic for the RS232 protocol) used to have a crystal that was resonating at 115200 Hz, meaning it could only produce baud rates that were dividable with that without a remainder. In other words, with divisor of 1 it gave 115200 baud, with 2 it gave 57600 Hz etc. There's no divisor for 31250. Now 3MHz/4Mhz is considerably bigger clock rate, which allows higher divisors and with that more precise baud rates, but still you has to divide it to get 31250 without a remainder (so if 3Mhz is actually 3000123 Hz, that could be a problem). I suppose it should work, but I haven't tried it.

Here's a good explanation on the acceptable error: https://www.allaboutcircuits.com/technical-articles/the-uart-baud-rate-clock-how-accurate-does-it-need-to-be/

But I think divisor 8 is extremely small, 3000000/31250 gives 96 as divisor (that is, if there are no stop bits or parity bits). To get how many bytes you can receive, you should calculate how many bits you can receive per sec (that's clock freq / divisor), and divide that by the number of bits per byte. That is not necessarily 8, because the UART can be configured to use start and stop bits, parity bits; and a stop bit can be even 1.5 bit in length. Hope it's clear, I'm not good at explaining this.

I'm not familiar with MIDI, but a quick search gave me this site https://www.compuphase.com/electronics/midi_rs232.htm which says you'll need a start bit and a stop bit too, meaning one byte would require 10 bits. It also suggests that you can't directly connect MIDI to an RS232 serial port, you'll need a small circuit (however it is possible you won't need this with the RPi, if you can configure baud rate correctly and the voltage level is 5V).

Cheers,
bzt

nickverlinden, 2019-12-13:
Hi bzt,

Thanks for helping me out. Now that I have set mbox[0] to 74, i get the clock speed back on the real hardware as well. This makes me able to verify that setting the clock speed in fact works.

You know, setting the clock to 3mhz, and the divider to 6 to get the 32150 baud rate, I do get data, and it seems like 3 bytes per key press on my midi keyboard. And that is correct, for every press there should be a 3 byte message. the bytes i'm getting are just wrong (on keypress the first byte should be 0x80). I'm starting to think this might have something to do with my octocoupler circuit. It might be wrong. To eliminate this problem I ordered a midi shield from china. That way I'm certain my circuit is not malfunctioning. I'm quite new to electronics, so I'm monkeying schematics I find on the web without verifying these are even correct.

If I have more news (it might take a while for the midi shield to arrive from China), I'll let you know. Would you find it useful if I contributed a midi example if I get it working?

Kind regards,
Nick

bztsrc, 2019-12-17:
Hi,

And what is the byte that you are receiving? Is the first byte 0x7F by any chance?
If so, then you should negate the bits because MIDI uses a different encoding (on RPi UART high voltage is 1, while on MIDI that's 0). Print out the bytes in binary bits, and you'll see.
```
10000000 <- should be this (0x80)
01111111 <- but if you receive this (0x7F) then you need to negate the bits.
```
(but again, I'm just guessing here, I have no experience with MIDI, I've just read the spec on the link above)

Cheers,
bzt