bztsrc/raspi3-tutorial #62: sd_writesector INT_WRITE_RDY timeout

DamianOslebo, 2020-01-01:
Great set of tutorials and happy new year! I was able to use your code to load some constant files on the SD into memory for some operations, but now I want to write an output file based off your last tutorial. However, your example times out for me at line 242 after the EMMC sends command 0x18220010 arg 1. The interrupt for INT_WRITE_RDY (0x00000010) is never set, and I have tried extending the timeout by a factor of 10. I am using a raspberry model 3B+, and a SanDisk Edge 32GB class A1 micro SDHC card. Thanks for any help.

bztsrc, 2020-01-06:
Hi,

Thank you very much and happy new year to you too!

I'm terribly sorry, but I'm afraid I can't help with that, because I don't have such an SD card. All I can say is that write ready should have been set. I'll try to get such a card and debug using it, but I can't promise anything. In the meantime, try another SD card, for others that always solved the write issues.

Cheers,
bzt

DamianOslebo, 2020-01-16:
Can you provide an sd card make and model that used to verify this example works? I can try whatever you are using. I have tried 5 different cards and two different pi's and they all timeout after the write cmd

bztsrc, 2020-01-16:
5 different cards you say? That's disturbing. Were they Class 10 SDHC cards? Can you use them for writing under raspbian?
I'm thinking hard how I could solve this. The problem is, this code works for me and I cannot fix an error that I can't reproduce. What about dumping the eMMC registers? Could you do that (use uart_hex() calls) and send me the results? Maybe I can figure it out from there, if there's another bit that toggles when the write operation is done.

Cheers,
bzt

DamianOslebo, 2020-01-16:
Thanks for the time to investigate. I can write files using Raspbian OS on the current card that I am using. I am compiling using gcc-arm-8.3-2019.03-i686-mingw32-aarch64-elf tools and all the same makefiles and code. I have tried with and without the timeout disabled on the EMMC_CONTROL1 register.

Current Raspberry Pis: Model 3B+, Model 3 vi. 2
Current SD cards: 32 GB SanDisk ImageMate A1, 16GB SanDisk Ultra 10/A1

Here is the register dump before the send write command:
EMMC_STATUS: 0x01FF0000
EMMC_CONTROL0: 0x00000002 
EMMC_CONTROL1: 0x000E0207
EMMC_CONTROL2: 0x00000000
EMMC_INTERRUPT: 0x00000002
EMMC_INT_EN: 0x37FF7FFF
EMMC_INT_MASK: 0x37FF7FFF
EMMC_BLKSIZECNT: 0x00010200
EMMC_SLOTISR_VER: 0x99020001

Here is the register dump after the interrupt timeout check before putting data into EMMC_DATA:
EMMC_STATUS: 0x01FF0206
EMMC_CONTROL0: 0x00000002 
EMMC_CONTROL1: 0x000E0207
EMMC_CONTROL2: 0x00000000
EMMC_INTERRUPT: 0x00108000
EMMC_INT_EN: 0x37FF7FFF
EMMC_INT_MASK: 0x37FF7FFF
EMMC_BLKSIZECNT: 0x00010200
EMMC_SLOTISR_VER: 0x99020001

Please let me know if you need anymore information.

bztsrc, 2020-01-17:
Thank you very much, this is very helpful! Yeah, EMMC_CONTROL hasn't changed for sure. But EMMC_INTERRUPT has, leave this with me, I'll check the spec to see what those bits are. Polling EMMC_INTERRUPT makes much more sense anyway.

Cheers,
bzt

DamianOslebo, 2020-01-21:
While you are investigating this issue, do you mind passing me the model information for the raspberry pi and SD cards that you have tested and can confirm work with your code? Thanks, I want to rule out any of my own user error issues in all of this.

bztsrc, 2020-01-23:
Hi,

I've repeated the test with a card. It's a Kingston MicroSDHC 8GB card made in Japan. There's some small letter on it which I believe could be the model, but it's so small I can really make it out. Should be something [8B][O0]C[8B][OG] I think. The last two letters are more likely "8G" than "BO".

Although I have to tell you that my Rasppi board was made in China (not entirely sure it was made in a verified factory). It could be it is my board that handles the cards differently. If I can found a combination that works for you and works for me, this could improve the value of these tutorials greatly.

Cheers,
bzt

DamianOslebo, 2020-01-24:
bzt,
I found this on amazon:
https://www.amazon.com/Kingston-Digital-8GB-Industrial-SDCIT/dp/B01DOFCPNW/ref=sr_1_4?keywords=kingston+microsd+8gb&qid=1579890295&sr=8-4
Does this match your card?

Also when you poll for sd_status on line 224 of sd.c you use constant SR_WRITE_AVAILABLE which you have as 0x00000400. It looks like from the bcm2735 documentation (page 72) that this value should be 0x00000100 instead (SR_READ_AVAILABLE should be 0x200). Was there a reason you chose this value instead? Changing SR_WRITE_AVAILABLE and SR_READ_AVAILABLE to correct value still does not fix the timeout problem for me.

DamianOslebo, 2020-01-24:
Found the problem!
#define CMD_WRITE_SINGLE    0x18220010
#define CMD_WRITE_MULTI     0x19220032
should be:
#define CMD_WRITE_SINGLE    0x18220000
#define CMD_WRITE_MULTI     0x19220022
TM_DAT_DIR was set from sd card to host when it should be host to card. Oh my gosh, I have beating my head on this for a month! Anyways great set of tutorials. I am going to start on the memory cacheing exercises now to improve the performance of my algorithm.

bztsrc, 2020-02-25:
Thank you very much for you patience. I've mentioned your contribution, very helpful, thanks!

Cheers,
bzt

closed 2020-02-25 by bztsrc