bztsrc/raspi3-tutorial #69: 0B_readsector GPIO47 High Event Interrupt Enabled

astrelsky, 2020-04-06:
https://github.com/bztsrc/raspi3-tutorial/blob/e2bcbff7cfe8c7152c938927b91c2ba00fcb6e5b/0B_readsector/sd.c#L257

This is enabling the GPIO High Detect Enable for GPIO 47 (activity led I think). Is there a reason for this?

bztsrc, 2020-04-09:
Huh, that was a long time ago... I must admit I don't remember all the bits. Enabling the activity LED should not cause any harm even if it's not blinking in lack of interrupts. Could be unnecessary, however if it works this way, then I rather not change the code.

I encourage you to test it with and without that bit set to see if there's any difference at all.

Nice catch, btw!
bzt

closed 2020-04-09 by bztsrc

astrelsky, 2020-04-09:
> Huh, that was a long time ago... I must admit I don't remember all the bits. Enabling the activity LED should not cause any harm even if it's not blinking in lack of interrupts. Could be [...]

If you intended to blink the activity led then you should be using `GPCLR1` to disable and `GPSET1` to enable it. What the code there is doing is enabling the High Event Interrupt. If someone were to use the 0B_readsector example for an sd card driver like https://github.com/boochow/micropython-raspberrypi then if they try to use irq 52, the interrupt which triggers for all gpio pins, they will encounter the interrupt always being triggered.

`GPCLR1` is missing from you gpio.h This would be the correct definition. `#define GPCLR1          ((volatile unsigned int*)(MMIO_BASE+0x0020002C))`