bztsrc/raspi3-tutorial #70: OD_readfile - An exception is triggered when trying to load FAT table

DarthBakerXD, 2020-04-25:
When i'm trying to read a file (kernel8.img) and load it from the SD card to the memory, it triggers an exception when it tries to load the FAT table. After some debugging, I noticed that the exception is triggerd after exactly 7050 sectors have been loaded from the SD card to the memory (which mean that when the exception is triggered , "c" variable equals to 7050). I think the reason behind this is that buffer address is clashing with other variable addresses (especially "num" - the total number of sectors to be loaded to the memory) so the buffer runs over the other variables. Even when I make the readblock function to skip other variable addresses, the exception is still triggered after 7050 sectors have been loaded. Here is an example:

[image]

For the readblock function parameters - lba is 2001 hexadecimal and num is 3B70 hexadecimal.
The exception is triggered when Return is called at the end of the function.
sd_err is zero, in the last line I printed the values of "c" variable, "num" variable and the buffer address variable. "c" is 7050 (like I said before), "num" is zero because it was run over by the buffer shortly before the exception was triggered, and the buffer address is 200 hexadecimal (512 decimal) since it became zero but then 512 decimal was added to the address as part of the while loop.

I have been trying to fix this for hours and although I know what the problem is (I think) I have no idea how to fix this.

Please help me if you can!

bztsrc, 2020-04-26:
Hi,

Thanks for the feedback!

> I think the reason behind this is that buffer address is clashing with other variable addresses

Unlikely, because variables are allocated in the bss section, and the buffer is allocated _after_ the bss section. But regardless the issue could be a corrupted variable, yes.

> lba is 2001 hexadecimal and num is 3B70 hexadecimal.

It was a long time ago when I wrote this tutorial, but I think 3B70 should be 3B50. I'll look into that, but right now I don't think that's the issue (but I could be wrong). Maybe loading the clusters overwrites the FAT in memory? (just a guess)

> Please help me if you can!

Try to decode ESR, that might help. Also it would be helpful if you could somehow upload the SD image so that I can recreate the issue. Can you upload the zipped image somewhere and post the link here?

Cheers,
bzt

bztsrc, 2020-04-26:
Note: ELR being 0x14 is definitely wrong, and means the code jumped outside of the text segment. Suggests stack corruption, as you assumed, but I can't see how that could happen.

bztsrc, 2020-04-30:
Any chance you could provide the disk image so that I can reproduce the error?

Cheers,
bzt

bztsrc, 2020-09-23:
Since I can't reproduce this, and you didn't provide an image to test with, I'm going to close this issue. If you can provide a way to consistently reproduce the problem, please reopen.

Cheers,
bzt

closed 2020-09-23 by bztsrc