bztsrc/raspi3-tutorial #76: Support for/compatibility with Raspberry Pi 4?

EdMckrete, 2020-09-21:
BCM2711 (Cortex-A72) vs BCM2837 (Cortext-A53) obviously the biggest difference.
But does the core difference or anything else in the RPi 3->4 delta impede use of this tutorial as is?

bztsrc, 2020-09-23:
Well, the biggest difference is, the MMIO is at different address. For RPi 3 (A53) that's 0x3F000000, while RPi4 (A72) uses 0xFE000000. You'll need to change [gpio.h at line 26](https://github.com/bztsrc/raspi3-tutorial/blob/master/03_uart1/gpio.h#L26).
As for the GPIO head mapping, it goes slight differently (meaning the serial port on pin 14/15 might not work for these tutorials), otherwise everything else (mailboxes, random generator, framebuffer etc.) is compatible.

I was thinking about creating a raspi4-tutorials version of this, however I didn't had the time for that yet. I've moved this now up on my TO DO list.

Cheers,
bzt

closed 2020-09-23 by bztsrc

imabuddha, 2020-12-27:
Just wanted to say that I recently got a Pi 4 & I'd be interested if you get around to making a raspi4-tutorials version. 🙇🏻

I'm also considering doing either the Stanford CS140e or Georgia Tech CS-3210 courses but on the Pi 4, so it'd be helpful for that as well.