bztsrc/raspi3-tutorial #80: Interrupts?

ajh123, 2020-10-15:
can we have an example for Interrupts or do we just give the mailboxes different parameters

bztsrc, 2020-10-16:
Yep, a tutorial on interrupts is still missing. I'll try to find time to write it!

In short, you'll have to program the interrupt controller, route IRQs to cores. Then you also have to tell the peripheral to generate an IRQ (this might happen with a Mailbox call). When IRQ is fired, the IC checks if it's enabled, and then routes it to one (or all) of the cores, which receive an interrupt. Then the CPU looks up VBAR, and calls a subroutine. In that subroutine you have to check which peripheral caused the interrupt (and depending on the peripheral you also might acknowledge).

Until the tutorial is written, [here](https://gitlab.com/bztsrc/osz/-/blob/master/src/core/aarch64/rpi3/intr.c)'s a example code.
intr_init() - disables all IRQs, and sets up ARM local timer
intr_getirq() - returns which IRQ triggered the interrupt
intr_enable()/intr_disable() - enable or disable an IRQ
clock_ack() - acknowledge IRQ in the ARM local timer

Hope this helps,
bzt

bztsrc, 2020-10-16:
Maybe this wasn't clear enough. What I wanted to say is, whether you can use Mailbox calls and whether you have to acknowledge the IRQ depends on the peripheral. For example the ARM local timer is enabled via MMIO, and it does need an acknowledgement (CLOCK_IRQ in the example). The per core timer on the other hand is enabled via system registers and does not need an acknowledgement (SCHED_IRQ in the example).

Cheers,
bzt

closed 2020-10-16 by bztsrc

ajh123, 2020-10-16:
Cheers, i will try to add it to [my os](https://github.com/ajh123/pyOs) thing,