bztsrc/raspi3-tutorial #84: I can't see anything in qemu even I added the -serial stdio.

xubenji, 2021-01-26:
Hi, bztsrc. I am a beginner in developing OS. I read the readme.MD in this tutorial and I have seen this question propose by your other fans. He got the same problem that is he can't see anything on the screen of qemu. You suggest him to add -serial stdio after that order. 
In the 03_uart1 lesson, I added those words, but I still can't see anything on the screen. Why? Here is my entire order:
qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial stdio

bztsrc, 2021-01-30:
Try running `make run`. There's a correct qemu invocation in all [Makefile](https://github.com/bztsrc/raspi3-tutorial/blob/master/03_uart1/Makefile.gcc). For 03_uart1, that is
```
qemu-system-aarch64 -M raspi3 -kernel kernel8.img -serial null -serial stdio
```
> I read the readme.MD in this tutorial

Try to pay more attention on what you read. For OSDev that's essential. Programming low-level hardware is a very tricky business, you must be careful. There's an answer in the readme to your question.

Cheers,
bzt

closed 2021-01-30 by bztsrc

xubenji, 2021-01-30:
> Try running `make run`. There's a correct qemu invocation in all [Makefile](https://github.com/bztsrc/raspi3-tutorial/blob/master/03_uart1/Makefile.gcc). For 03_uart1, that is
>
> ``` [...]

Thanks for your response. I can see "hello world" in the terminal, but I can't see anything on the qemu screen. The screen is black.

This is

bztsrc, 2021-02-01:
> I can see "hello world" in the terminal, but I can't see anything on the qemu screen.

This is perfectly normal, because we haven't printed anything on screen. In qemu, you should see a black window, on real hardware the rainbow splash or black screen.

It is the 09_framebuffer tutorial that first draws something at all on screen, and 0A_pcscreenfont that displays text.

Cheers,
bzt