bztsrc/raspi3-tutorial #85: No input with UART1

Yul3n, 2021-02-06:
I built and ran the third tutorial on UART1, output works fine, I can see "Hello World!". However, when I type something nothing happens.
There is an infinite loop in the function `uart_getc`, on the line 75: `do{asm volatile("nop");}while(!(*AUX_MU_LSR&0x01));`
I'm using qemu.
Thanks!

bztsrc, 2021-02-06:
Hi,

> There is an infinite loop

That's no infinite loop. It stops as soon as the AUX status register reports byte received (AUX_MU_LSR bit 0 set).
FYI, I'd suggest to move on to UART0, as the PL011 chip is more like the 16550 chip on the original IBM PC. It works better and it supports break signal, etc.

Cheers,
bzt

closed 2021-02-06 by bztsrc

Yul3n, 2021-02-06:
Thanks, yes actually even though I wrote things on my terminal, it didn't report anything