bztsrc/raspi3-tutorial #87: Explanation for mailboxes

Tix3Dev, 2021-05-08:
Hi,

I am working on an OS for the rpi 3 b+ with my friend. This repo is a very good resource if we don't get the hang out of all of the docs. So first of all thank you very much for that.

We are currently struggeling a bit with the mailboxes, so we looked at your code.
```
    // get the board's unique serial number with a mailbox call
    mbox[0] = 8*4;                  // length of the message
    mbox[1] = MBOX_REQUEST;         // this is a request message
    
    mbox[2] = MBOX_TAG_GETSERIAL;   // get serial number command
    mbox[3] = 8;                    // buffer size
    mbox[4] = 8;
    mbox[5] = 0;                    // clear output buffer
    mbox[6] = 0;

    mbox[7] = MBOX_TAG_LAST;
```
Earlier we looked at this: https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#tags-arm-to-vc

We see the connection between the docs and your code only for a few points, namely [0], [1], [2] and [3].
But what comes after?

We would be very thankful for an explanation so we could make the arrays just by looking at this site: https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#tags-arm-to-vc

bztsrc, 2021-05-10:
Correction, on that page at section "Tag format", it still says that [4] should be the output value's length.
```
Tag format:

    u32: tag identifier                          <-- [2]
    u32: value buffer size in bytes      <-- [3]
    u32:                                      <-- [4]
        Request codes:
            b31 clear: request
            b30-b0: reserved
        Response codes:
            b31 set: response
            b30-b0: value length in bytes
    u8...: value buffer                 <-- [5]..
```

Tix3Dev, 2021-05-11:
Thank you for your response.

I think we mostly understand it now. Today we just made the repository of our project public so you are welcome to look at it (and maybe also look at the mailbox code): https://github.com/GENADEV/GENADEV_OS

Something that confused us was the fact, that this tag https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-board-model returned 0 as result. So first we thought something isn't working 😅 (or is it really not working?).

bztsrc, 2021-05-15:
TBH I've never used that mailbox. I've always used the MIDR system register. See [here](https://wiki.osdev.org/Detecting_Raspberry_Pi_Board)

Cheers,
bzt