bztsrc/raspi3-tutorial #92: Please help to understand the load address 0x800000 in the linker file

drymon, 2021-07-22:
Hi @bztsrc ,

I greatly admire and appreciate the effort you put into creating and sharing this project. 
There's something I don't quite understand, could you please teach me?
I see in your link.ld file containing load address .=0x80000 but in similar file at:
https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/linker .ld 
I don't see it.
Can you please clarify this difference for me?

bztsrc, 2021-07-23:
Hi,

There are different operating modes on the RPi. My examples use the default memory mapping, where the kernel is loaded at 0x80000 (also meaning no config.txt needed).

Matyukevich's tutorial has a [config.txt](https://github.com/s-matyukevich/raspberry-pi-os/blob/master/src/lesson01/src/config.txt) file, and in it there's a "kernel_old=1" line. That makes the kernel to be loaded at address 0, therefore there's no need to set the address in the linker script (as it defaults to 0). The downside is, you'll need the compiled kernel8.img AND a special config.txt too on the USB stick's boot partition. Also keep in mind that the kernel_old option might be obsoleted some day.

There are other differences with kernel_old, but the load address is what influences the linker script.

Cheers,
bzt

closed 2021-07-23 by bztsrc

drymon, 2021-07-23:
I understand now, thank you very much.