bztsrc/raspi3-tutorial #95: How do you debugger with gdb-multiarch?

xubenji, 2021-09-19:
Good morning, Zoltan Baldaszti
I was learning your tutorial for a long time. today, I want to debugger your kernel I got a problem that is gdb-multiarch can't load the symbol file kernel8.elf. When I load the symbol file, the gdb-multiarch gave me this information. 
```

(gdb) symbol-file kernel8.elf
Reading symbols from kernel8.elf...(no debugging symbols found)...done.
```
I have set the architecture to aarch64 for sure. Do you know which mistake to occur this problem?

xubenji, 2021-09-19:
the strange thing is that I can debugger the other kernel running on the X64 as well. Could you help me to find the problem?

Chronal, 2021-09-20:
I briefly checked the Makefiles, the `CFLAGS` variable do not have `-g` flags. This means that binaries compiled will not have debugging symbols in them.

For a nice experience in gdb, remove `-O2` and add `-g` to the `CFLAGS` variable in the Makefile of whichever section you wish to experiment with.

bztsrc, 2021-09-20:
Yes, that's it. You have to add `-g` to the command line flags to have debug symbols. Removing optimizations (or using `-O0`) is also a good advice, because then the compiled code will follow the source more (not inlining parts, splitting functions, etc.).

Just for the records, if you don't use "strip" on the elf, then you can still use gdb with function and variable name symbols. So for example, despite of the above warning, you can use the "main" symbol, and gdb will print out "main+0x1234" offsets for example, but with `-g` it will print the name of the source file and line too like "function main in main.c:12" which is much more helpful. Adding debugging symbols can increase your kernel's size considerably, so only use `-g` when you're debugging.

Cheers,
bzt

closed 2021-09-20 by bztsrc

kgdngitfk, 2023-11-25:
hi I set CFLAGS = -Wall -g -ffreestanding -nostdinc -nostdlib -mcpu=cortex-a53+nosimd ;then make all, use GDB like this
`
(gdb) symbol-file kernel8.elf
can't read symbols: file format not recognized`
how to solve it ,plz

kgdngitfk, 2023-11-26:
or can you make more details about how to use qemu and gdb to debug the kernel ,which is import for the  newbie as a tutorial.
and why should write  code like this  reference to the BCM2837 documents and other docs ,such as :
```
 //read cpu id, stop slave cores
mrs x1, mpidr_el1    //it's confusing why read the mpidr_el1 can get the cpu id
and     x1, x1, #3
cbz     x1, 2f 
```
according to arm official document : Affinity level 0. This is the affinity level that is most significant for determining PE behavior. Higher affinity levels are increasingly less significant in determining PE behavior. The assigned value of the MPIDR.{Aff2, Aff1, Aff0} or [MPIDR_EL1](https://developer.arm.com/documentation/ddi0595/2021-12/AArch64-Registers/MPIDR-EL1--Multiprocessor-Affinity-Register?lang=en).{Aff3, Aff2, Aff1, Aff0} set of fields of each PE must be unique within the system as a whole.
there seems no description about the cpu id,
but according to your code i guess that the cpu ids are 0, 1, 10, 11(binary) ; so that there is and x1,x1,#3
and so on
thanks

xubenji, 2023-11-26:
> hi I set CFLAGS = -Wall -g -ffreestanding -nostdinc -nostdlib -mcpu=cortex-a53+nosimd ;then make all, use GDB like this ` (gdb) symbol-file kernel8.elf can't read symbols: file format not recognized` how to solve it ,plz

I guess you may need to use the 

> gdb-multiarch