bztsrc/raspi3-tutorial #96: Why ptr+=4 or ptr+=3?

xubenji, 2021-09-26:
In 09_framebufer, lfb.c, I found that for each pixel,  you move pointer ptr 4bytes. You said the image's pixel is RGB. I think that the pixel in RGB format should take over at 3 bytes. Why do you move 4bytes each time?
` for(y=0;y<homer_height;y++) {
        for(x=0;x<homer_width;x++) {
            HEADER_PIXEL(data, pixel);
            // the image is in RGB. So if we have an RGB framebuffer, we can copy the pixels
            // directly, but for BGR we must swap R (pixel[0]) and B (pixel[2]) channels.
            *((unsigned int*)ptr)=isrgb ? *((unsigned int *)&pixel) : (unsigned int)(pixel[0]<<16 | pixel[1]<<8 | pixel[2]);
            ptr+=4;
        }
        ptr+=pitch-homer_width*4;
    }`

bztsrc, 2021-09-26:
Hi,

> You said the image's pixel is RGB. I think that the pixel in RGB format should take over at 3 bytes.

Nope, pixel depth is 32 bits, 4 bytes. See [source](https://github.com/bztsrc/raspi3-tutorial/blob/master/09_framebuffer/lfb.c#L59).

There's an extra byte for alignment. One pixel takes 32 bits, that is 8+8+8 bits for RGB, and another 8 bits unused. Without MMU, you can't access dwords which are not 4 bytes aligned, you would get an alignment fault. Furthermore the framebuffer isn't normal memory, rather MMIO, where aligned dword access has an optimized implementation in the hardware circuit. It is much much faster than accessing RGB bytes one-by-one, so you should write a dword at once, but that requires proper alignment.

Cheers,
bzt

closed 2021-09-26 by bztsrc

xubenji, 2021-09-26:
Hi, bztsrc.
Thanks for your response. By the way, how do you change the pixel to 3 bytes each one, instead of 4bytes?

bztsrc, 2021-09-26:
> By the way, how do you change the pixel to 3 bytes each one, instead of 4bytes?

Use 24 bits as pixel depth in the mailbox property. But as I've pointed out, don't do that, because you won't be able to handle that framebuffer in any meaningful way. Accessing by bytes is extremely slow (there's a hardware overhead when the VC refreshes, and by changing colour channels byte by byte you'll get 3 times overhead plus bad colour artifacts), and there's no other way to access unaligned 24 bits at once. Accessing the first pixel at offset 0 with unsigned int would overwrite the blue (or red) component of the second pixel, and accessing the second pixel at offset 3 would result in a CPU misaligned access exception.

That is, if the VC firmware even supports 24 bits per pixel, most modern video hardware only supports 16 or 32 bits (I haven't tried with the latest RPi firmware).

Cheers,
bzt

xubenji, 2021-09-27:
Thank you very much!

xubenji, 2021-09-27:
Hi, bztsrc
At the start of this tutorial, you use uart0 and uart1 to send the alphabet to the terminal instead of the screen. I want to send something to the screen instead of the terminal without the graphic mode. Do you have any advice?

bztsrc, 2021-09-27:
> without the graphic mode

I don't think that's possible. All modern video cards and GPUs (RPi's VC included) supports graphics mode only, nothing else. There's no teletype text mode any more like old EGA/VGA cards used to have. If you want to display a character on screen, your only option is rendering fonts.

See the [0A_pcscreenfont](https://github.com/bztsrc/raspi3-tutorial/tree/master/0A_pcscreenfont) tutorial. An easy way out is using a header only library to do this for you, like [Scalable Screen Font](https://gitlab.com/bztsrc/scalable-font2). I would not recommend the freetype2 library, as it is extremely hard to integrate in a bare metal project due to its dependencies. Use a dependency-free, stb-style header only lib instead, or copy'n'paste the print function from the 0A_pcscreenfont tutorial.

PC Screen Font is pretty easy to handle, but it won't give you full UNICODE support (which is ok if you're only interested in English-only messages). On the other hand SSFN supports full UNICODE and UTF-8, but it is more complex. I provide a simplistic print function for both in the tutorial. It depends on your needs which one to choose.

Cheers,
bzt