bztsrc/raspi3-tutorial #102: 0x1C2 no longer has 0xE or 0xC partition type

codernavi18, 2022-03-23:
I am trying to develop my FAT32 disk image. I followed the steps from here : https://github.com/procount/fat32images/blob/master/createfat32. Tried both `mkfs.fat` and `mkfs.vfat` but still the `0x1C2` remains set to 0. In the raspbian FAT image, this is perfectly set to 0xE. 

Are there any recent changes in FAT filesystem as why this byte at `0x1C2` is no longer set? How did you prepare the disk image?

Below is my hexdump of a FAT file with above steps : 
```
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001b0  00 00 00 00 00 00 00 00  d7 76 46 6c 00 00 00 82  |.........vFl....|
000001c0  03 00 0c 8a 08 82 00 20  00 00 00 e0 1f 00 00 00  |....... ........|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 55 aa  |..............U.|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00400000  eb 58 90 6d 6b 66 73 2e  66 61 74 00 02 08 20 00  |.X.mkfs.fat... .|
00400010  02 00 00 00 00 f8 00 00  3f 00 ff 00 00 00 00 00  |........?.......|
00400020  00 e0 1f 00 f8 07 00 00  00 00 00 00 02 00 00 00  |................|
00400030  01 00 06 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00400040  80 00 29 8d 75 8d 46 52  45 43 4f 56 45 52 59 20  |..).u.FRECOVERY |
00400050  20 20 46 41 54 33 32 20  20 20 0e 1f be 77 7c ac  |  FAT32   ...w|.|
00400060  22 c0 74 0b 56 b4 0e bb  07 00 cd 10 5e eb f0 32  |".t.V.......^..2|
00400070  e4 cd 16 cd 19 eb fe 54  68 69 73 20 69 73 20 6e  |.......This is n|
00400080  6f 74 20 61 20 62 6f 6f  74 61 62 6c 65 20 64 69  |ot a bootable di|
00400090  73 6b 2e 20 20 50 6c 65  61 73 65 20 69 6e 73 65  |sk.  Please inse|
004000a0  72 74 20 61 20 62 6f 6f  74 61 62 6c 65 20 66 6c  |rt a bootable fl|
004000b0  6f 70 70 79 20 61 6e 64  0d 0a 70 72 65 73 73 20  |oppy and..press |
004000c0  61 6e 79 20 6b 65 79 20  74 6f 20 74 72 79 20 61  |any key to try a|
004000d0  67 61 69 6e 20 2e 2e 2e  20 0d 0a 00 00 00 00 00  |gain ... .......|
004000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

bztsrc, 2022-03-24:
> Are there any recent changes in FAT filesystem as why this byte at 0x1C2 is no longer set?

No. The file system on a partition has nothing to do with the MBR tables, those have to be set independently (this was always like this, so no change here).

Long story short, what you're doing is a single file system image. That alone won't work, because on a typical drive you can have multiple file systems, each in its own partition. So the very first sector of a drive is not a VBR (Volume boot record, the first sector of a FAT file system), rather an MBR (Master boot record) with a partitioning table. There each partition has a type, a first sector address and a number of sectors field. The file system only occupies the area pointed by that partition table entry, it is the first sector of the partition that should have the VBR.

If you're on Linux, then you can use the `fdisk` command to create a partitioning table and set the file system type of a partition for example. Then you can set up a loopback device which points inside that disk image to the desired partition area, and then you can use mkfs.vfat to create a file system on that loopback device. See more details [here](https://wiki.osdev.org/Bootable_Disk) (note: this tutorial creates a GPT table, you'll need an MBR table, so do not use the `g` command, rather use the `o` command).

If you want to do this programmatically, [here](https://gitlab.com/bztsrc/bootboot/-/blob/master/aarch64-rpi/mkboot.c) is a little a tool I wrote that locates the ESP in GPT and then mirrors it into the MBR partitioning table. You can skip the GPT part, just take a look at the MBR part. But you'll still have to create and add the partition image.

Cheers,
bzt

closed 2022-03-24 by bztsrc

bztsrc, 2022-03-24:
Maybe I wasn't clear, when you do
```
naveen@workstation:/tmp$ mkfs.fat disk.img
```
That will overwrite whatever you have in the img file with a single file system. What you need is, set up a loopback device which points inside the disk.img, and then use
```
naveen@workstation:/tmp$ mkfs.fat /dev/loop0
```

One more thing to keep in mind: your image is just 51Kb. You cannot create a valid FAT32 on that. FAT file system has a minimum storage space requirement of 33Mb. See [size limits](https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system#Size_limits). I suggest to use FAT16 (RPi firmware can handle that too just fine), `mkfs.fat -F 16`. The minimum size for FAT16 is just 2Mb.

Cheers,
bzt