bztsrc/raspi3-tutorial #104: Typos - bareminimum readme

greenonline, 2022-07-01:
There are a couple of typos in the [readme.md](https://github.com/bztsrc/raspi3-tutorial/tree/master/01_bareminimum#readme) for 01-bareminimum:

- "clearity" should be "clarity"
- "preambule" is French, in English it is "Preamble"

---

Nice tutorial by the way. Thanks for the work. :-)

bztsrc, 2022-07-02:
Thank you, typos have been fixed!

closed 2022-07-02 by bztsrc

greenonline, 2022-07-09:
Sorry, there is another (minor) typo, that I hadn't noticed before, on the same page. Under the **Makefile** section, the word "finaly" should be "finally" - there are two "l"s.

Sorry for not noticing it before - I was busy concentrating on getting the build tools working on part 00.