bztsrc/raspi3-tutorial #117: about the uart

kgdngitfk, 2023-11-30:
why don't we need operation the GPIO high/low generate the pulse ( for example 0100 0001 of  charactor A) to implement the uart data transportation . and we just use:
 ```  void uart_send(unsigned int c) {
    /* wait until we can send */
    do{asm volatile("nop");}while(*UART0_FR&0x20);
    /* write the character to the buffer */
    *UART0_DR=c;
}
```

kgdngitfk, 2023-12-01:
the asscii of charactor A is  0100 0001, I think we should set a GPIO low/high as follow pseudocode
```
GPIO(PIN,LOW);
delay(10ms);
GPIO(PIN,HIGH);
GPIO(PIN,LOW);
delay(50ms);
GPIO(PIN,HIGH);
```
to send the data by the waveform，but we just      *UART0_DR='A'; that's why?

bztsrc, 2023-12-04:
Ah, I see.

TL;DR: because that would require to allocate the CPU entirely to the
serial communication, you couldn't do anything else (if you've tried,
you would loose precise timing which is essential for serial
communication).

Long version: we write to UART0_DR because behind the MMIO register
there's a <a href="https://documentation-service.arm.com/static/5e8e36c2fd977155116a90b5">PL011
chip</a>. This is an UART chip, which has a crystal that oscillates at
a given frequency (independent to the CPU clock). It also has a
register, which holds a byte (actually it's a FIFO with multiple
bytes). Now this chip's job is to check the bits of that byte
(independently to the CPU), and trigger the GPIO lines at the given
precise frequency, so you don't have to do all of that yourself. (In a
nutshell, in reality there's lot more to it, parity bits, stop bits
etc. synchronization and automatic error correction, modem-control
etc.)

For example on a typical desktop PC you'll find a 16550A chip (or
something that emulates that chip these days), which is a similar
UART. All these chips, all of them work the same way: you write a byte
into its register, and they do the rest in the background. You don't
have to waste CPU cycles on figuring out the exact delays, and your
program won't "freeze" until the entire byte is sent etc. Your CPU is
just responsible to write to the UART's register (one CPU cycle), and
then it can go on with whatever it was doing. The low-level
synchronization is taken care for you by the UART chip asynchronously
and automatically in the background.

kgdngitfk, 2023-12-05:
thanks, you are helpful and patient  to greenhand 👍

closed 2023-12-05 by kgdngitfk